sm0svx/svxlink #629: enhancement / question for an separate RX_ID

DG1FI, 2023-06-19:
Hi!

The goal is, to add a diffent roger peep for HF and the ReflectorLogic. The simple way is to define
under  [RX1] the RX_ID=R or some other letter and then modifiy the send_rgr_sound tcl procedure:

proc send_rgr_sound {} {
    variable sql_rx_id;

    if {$sql_rx_id == "R"} {

   playTone 2000 100 50

    } else {

    playTone 2000 100 50
    playSilence 150
    playTone 2000 100 50

   }
    set sql_rx_id "?";
}

Problem here: after the repeater open, the $sql_rx_id variable is set to "?"... so the beep will play two times after
the repeater open ident.

1. Is there any variable in that namespace that says that the repeater is closed?
2. It's possible to add RX_ID=REFL or someting else to ReflectorLogic? That makes my goal very easy.

73 Mario

sm0svx, 2023-07-18:
Does commit f768d3c2805228bdbd2819eac9bbcaaa5fcd69b7 help?

DG1FI, 2023-08-01:
Don't know how difficult it is, but really good if the Reflektor has an own RX_ID like in the [RX] Section