sm0svx/svxlink #630: 1.7.99.84 and .85

sm5gxq, 2023-07-19:
Hi Tobias,

I was very quick to update but there are a few questionmarks.

Using OPEN_SQL_FLANK=CLOSE, it seems two things are trying to happen at the same time; opening of the repeater and activation of a talk group. I got reports from Gothenburg, that the repeater did not activate the talk group, just opened - whatever mapped subtone was used. I have now set the CTCSS_TO_TG_DELAY somewhat lower than OPEN_ON_CTCSS to be on the safe side (in case the user does not send for the required time). I cannot test this myself, but I got reports that it did not always work. When looking at the log file, I can see that it works, but only some times, and some times not. I have instructed the user to always hold PTT down for the required time.

In the function squelch_open I had a CW::play, and this has worked perfectly before. But now, the repeater will send this morse letter, even if the repeater is not up. Well, it could be "by design" - and I have put a test for $repeater_is_up, to circumvent this.

I also noticed some strange behaviour from the remote trx, the last day. It gets frequent Heartbeat timeouts, but this might of course be unrelated. There is a microwave link between the sites.

At the same time, I have seen lots of APRS server timeouts. I will disable that, since it's not needed actually.

It's a bit hard for me to try a roll-back, since I have some changes of my own to RepeaterLogic.cpp :-).

rgds
Peter SM5GXQ

sm5gxq, 2023-07-19:
This is a good example:

Jul19 20:04:58.060: Voter[Rx1]: The squelch is OPEN (siglev=41)
Jul19 20:04:58.061: Voter[Rx2]: The squelch is OPEN (siglev=79)
Jul19 20:04:58.061: Voter: The squelch is OPEN (Rx2[Rx2:CTCSS Rx2:CTCSS-B Rx2:SIGLEV*73]=79)
Jul19 20:04:58.081: RepeaterLogic: 88.5 Hz CTCSS tone detected
Jul19 20:04:59.190: Voter[Rx1]: The squelch is CLOSED (siglev=-4)
Jul19 20:04:59.193: Voter[Rx2]: The squelch is CLOSED (siglev=-2)
Jul19 20:04:59.193: Voter: The squelch is CLOSED (Rx2[Rx2:CTCSS-B Rx2:CTCSS=88.5:3 Rx2:SIGLEV=-2]=-2)
Jul19 20:04:59.193: RepeaterLogic: Repeater UP, reason: CTCSS_CLOSE
Jul19 20:04:59.193: RepeaterLogic: Last ID too close, skipping ident
Jul19 20:04:59.193: Tx1: Turning the transmitter ON
Jul19 20:04:59.852: RepeaterLogic: Playing RGR sound
Jul19 20:05:27.175: Voter[Rx2]: The squelch is OPEN (siglev=28)
Jul19 20:05:27.185: Voter: The squelch is OPEN (Rx2[Rx2:CTCSS-B Rx2:CTCSS=88.5:3 Rx2:SIGLEV*36]=28)
Jul19 20:05:27.197: Voter[Rx1]: The squelch is OPEN (siglev=4)
Jul19 20:05:27.197: ReflectorLogic: Selecting TG #240602
Jul19 20:05:27.389: Voter[Rx1]: The squelch is CLOSED (siglev=0)
Jul19 20:05:27.531: Voter[Rx2]: The squelch is CLOSED (siglev=0)
Jul19 20:05:27.531: Voter: The squelch is CLOSED (Rx2[Rx2:CTCSS-B Rx2:CTCSS=156.7:-1 Rx2:SIGLEV=-2]=0)
Jul19 20:05:27.531: RepeaterLogic: Playing RXID: K
Jul19 20:05:28.755: RepeaterLogic: Playing RGR sound
Jul19 20:05:56.567: ReflectorLogic: Selecting TG #0
Jul19 20:05:58.107: RepeaterLogic: Repeater DOWN, reason: IDLE
Jul19 20:05:59.012: Tx1: Turning the transmitter OFF

This is from SK6RFQ, with no cpp code mods. MUTE_FIRST_TX_LOC=1 but after his second tx, the repeater should activate tg 240. The time for tg activation is set to 900 ms and time for open of repeater is 1000. Instead the repeater activates its default tg which is 240602.

/Peter

sm0svx, 2023-07-19:
Hmmm yes... There seem to be a couple of strange things in the new code. I'll have a look at it.

sm0svx, 2023-07-19:
And oh, set CTCSS_TO_TG_DELAY=0. The TG will not be activated until the repeater is up anyway.

sm0svx, 2023-08-01:
@sm5gxq, feed-back on this issue?

sm5gxq, 2023-08-01:
Hi,

Your fix for tg activattion is indeed working. Thanks.

My other question regarding squelch_open, is this now by design? I have edited my script to check for repater_is_up as a workaround.

Regarding heartbeat timeouts, they seem to have been a temporary problem.

Finally, regarding aprs, I haven't checked that yet.

/Peter

commit a9ed8ef3 (2023-08-09): Squelch_open event handler called while not rpt up

The Logic::squelch_open TCL event handler was called on squelch close
even though even though the repeater was not up in a RepeaterLogic.

Fixes #630
closed 2023-08-09 by sm0svx