sm0svx/svxlink #633: Reflector HTTP request parsing problem

dl1hrc, 2023-07-30:
I feel there is a problem in parsing http requests. I've detected the problem in AsyncHttpServerConnection class, method handleStartLine and it depends on the system/platform:
```
void HttpServerConnection::handleStartLine(void)
{
  std::istringstream is(m_row);
  std::string protocol;
  if (!(is >> m_req.method >> m_req.target >> protocol >> std::ws))
  {
    std::cerr << "*** ERROR: Could not parse HTTP header" << std::endl;
    disconnect();
    return;
  }
...
```
The line **_if (!(is >> m_req.method >> m_req.target >> protocol >> std::ws))_** fails on my Ubuntu and give back "TRUE" on my Raspbian systems.

my versions:
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)  (working without a problem)
gcc version 12.2.0 (Ubuntu 12.2.0-17ubuntu1) (*** ERROR: Could not parse HTTP header)

If I remove **_>>std::ws_** from the line it's working on Ubuntu. I found this thread https://stackoverflow.com/questions/63299651/stdws-behavior-on-eof
Seems that something has changed here depending on version?

Mni 73s de Adi / DL1HRC

sm0svx, 2023-08-01:
Yes that seem to be a bug. What if you apply the following patch. Does that work on both your platforms?
```
--- a/src/async/core/AsyncHttpServerConnection.cpp
+++ b/src/async/core/AsyncHttpServerConnection.cpp
@@ -341,7 +341,7 @@ void HttpServerConnection::handleStartLine(void)
 {
   std::istringstream is(m_row);
   std::string protocol;
-  if (!(is >> m_req.method >> m_req.target >> protocol >> std::ws))
+  if (!(is >> m_req.method >> m_req.target >> protocol) || !is.eof())
   {
     std::cerr << "*** ERROR: Could not parse HTTP header" << std::endl;
     disconnect();
@@ -359,7 +359,7 @@ void HttpServerConnection::handleStartLine(void)
   is.clear();
   is.str(protocol.substr(5));
   char dot;
-  if (!(is >> m_req.ver_major >> dot >> m_req.ver_minor >> std::ws) ||
+  if (!(is >> m_req.ver_major >> dot >> m_req.ver_minor) || !is.eof() ||
       (dot != '.'))
   {
     std::cerr << "*** ERROR: Illegal protocol version specification \""
```

sm0svx, 2023-08-08:
Resolved by commit 5a9971480005d364460d44e4bcb831dd04027865

closed 2023-08-08 by sm0svx