sm0svx/svxlink #642: Change roger beep volume

Manu-on7fq, 2023-10-01:
Hi,

Is it possible to change the roger beep volume ?

73's

Manu - ON7FQ

f5vmr, 2023-10-02:
All tones can be adapted or changed.

Find the .tcl files at /usr/share/svxlink/events.d/local and have a look. The tone frequency, duration of tone and audible volume in the command playTone are all adjustable.

Regards
Chris G4NAB

Manu-on7fq, 2023-10-02:
Hi Chris,I have already found it under usr/share/svxlink/events.d/ Logic.tcl## Executed when the squelch have just closed and the RGR_SOUND_DELAY timer has# expired.#proc send_rgr_sound {} {  variable sql_rx_id  if {$sql_rx_id != "?"} {    # 150 CPM, 1000 Hz, -4 dBFS    CW::play $sql_rx_id 150 1000 -4    set sql_rx_id "?"  } else {    playTone 440 500 100  }  playSilence 100}Still, thanks for the tip!73sManu

closed 2025-03-16 by sm3sgp