sm0svx/svxlink #647: METAR

sm5gxq, 2023-10-25:
Hi,

There seems to be a change at aviationweather.gov, so the "adds" resource is no longer available.

I have made these mods to /etc/svxlink/svxlink.d/ModuleMetarInfo.conf:

```
LINK=/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&hoursBeforeNow=3&stationString=
#LINK=/adds/dataserver_current/httpparam?dataSource=metars&requestType=retrieve&format=xml&hoursBeforeNow=3&mostRecent=true&stationString=
```

Further, it seems that the man file for ModuleMetarInfo.conf is missing?

rgds
Peter

f5vmr, 2023-10-25:
Yes, these changes made an awful lot of apps fail to work. I made the
changes in svxlink, but the flight simulator world is having to make an
effort. I wonder how much it affected everyone in aviation?

DH3PAE, 2023-11-10:
Hallo, my Name ist Stefan and today i have start the Weather Modul but it does not work.
I have change for 3 Weeks an with this String from Peter it have worked fine:
LINK=/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&hoursBeforeNow=3&stationString=
But today there is another change in the data server on avianweather.gov.
Ther is no entry with beginning Link=cgi    today it beginns with   /api/data
So what must I do now, for hearing the Flight Weather Information, Peter can you Help sm5gxq this is dh3pae..please contact me under wolken2019@web.de if you find the mistake. Thank you for all Help Best Greetings Stefan

f5vmr, 2023-11-10:
> So what must I do now, for hearing the Flight Weather Information, Peter can you Help sm5gxq this is dh3pae..please contact me under ***@***.*** ***@***.***> if you find the mistake. Thank you for all Help Best Greetings Stefan

AviationWEATHER.GOV HAS AN UPSTREAM Network Issue on their own website. It may resolve itself later.

Chris

DH3PAE, 2023-11-10:
Ok Thank your fort he Information Chris, so i will be wait fort he next time and write here back when the problem has solved.

73 de Stefan dh3pae

 

http://eichenwetter.homelinux.net

or

 

http://eichenwetter.homelinux.net/pwsWD/index.php

 

you can connect me via Echolink /Svxlink  Node Number 528305

f5vmr, 2023-11-10:
AviationWeather is now back delivering Metar

DH3PAE, 2023-11-10:
Thanks for Info

 

-DH3PAE-

circool, 2024-04-01:
hi. 
SvxLink v1.8.0@24.02-3-gcde00792
in ModuleMetarInfo.conf:
```
[ModuleMetarInfo]
NAME=MetarInfo
ID=5
TIMEOUT=120
TYPE=TXT
SERVER=https://aviationweather.gov
LINK=/cgi-bin/data/metar.php?hours=0&order=id%2C-obs&sep=true&ids=
STARTDEFAULT=UUEE
AIRPORTS=UUEE,UUDD,UUWW
```
 
run metar module:
```
ECHO4: Frankfurt, Germany

SimplexLogic: digit=5
SimplexLogic: digit=#
SimplexLogic: Activating module MetarInfo...
https://aviationweather.gov/cgi-bin/data/metar.php?hours=0&order=id%2C-obs&sep=true&ids=UUEE
Tx1: Turning the transmitter ON
./5: line 1:  4600 Segmentation fault      sudo -u svxlink svxlink
```

> cat /tmp/metar
```
UUEE 312330Z 11003MPS CAVOK 09/06 Q1012 R24C/CLRD62 NOSIG
```

How to be?

circool, 2024-04-01:
XML with 
```
LINK=/cgi-bin/data/dataserver.php?requestType=retrieve&dataSource=metars&hoursBeforeNow=3&stationString=
```
is not problem

dl1hrc, 2024-04-01:
Hi,

my last message doesn't seem to have been sent:

yes, the metar data returned has a different structure. When I developed
this many years ago the structure was like this:

   2009/04/07 13:20
   FBJW 071300Z 09013KT 9999 FEW030 29/15 Q1023 NOSIG

The advantage is that in this structure it can be recognized whether the
data is current. Without the date, you can only guess from the first
sequence of numbers (FBJW 07xxxxZ ....). But regardless, it shouldn't
lead to a segfault. I'll see what the problem is.

Mni 73s de Adi / DL1HRC

dl1hrc, 2024-04-01:
I've changed something and it has been checked in into my private repository. It would be very nice if some people could test it before I make a pull request. If you have some time please check out it by 
git clone https://github.com/dl1hrc
Please notice the small change in the manual page and the changes in ModuleMetarInfo.conf
Mni 73s de Adi / DL1HRC

dl1hrc, 2024-04-23:
Does it work or not?

dl1hrc, 2024-04-23:
ok, but there is more than just a changed link:
https://github.com/dl1hrc/svxlink/commit/6687881b8544a251af38b58535ec0135ad799353

f5vmr, 2024-04-23:
Ok, but I do not understand the issue. The ICAO and the current setup from SM0SVX has always worked. It has just been the link that changed.
The current build on my repeater work very well. This is recorded off air moments ago.

Chris
￼

dl1hrc, 2024-04-24:
There was a segfault in the case of a message that did not correspond to the expected structure, so I changed the way aviationweather responded slightly. Maybe someone can test this version (for example the Metar info from Moscow with the specific runway information) as I can't test all airports on my system
Mni 73s de Adi / DL1HRC.

f5vmr, 2024-04-24:
Wow, Ok, I’ve not encountered that, although I have noticed that after a cold start of MetarInfo where EGNT - Newcastle is the DEFAULTAIRPORT having announced all the weather the announcement continues - Met Report Not Valid…….. I suppose that this might be a manifestation of the issue with.txt I haven’t checked it myself. I might pull your code to make another effort. I have just completed a full bash script build from menu for English (GB & US), French and Spanish. It would be nice to make it absolutely perfect.

Chris