sm0svx/svxlink #649: Compile svxlink on Debian v12 Bookworm

sp2ong, 2023-11-02:
Compile current svxlink sources on Debian v12 Bookworm, show follwoing warning during compile:

-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Checking for module 'sigc++-2.0'
--   Found sigc++-2.0, version 2.12.0
-- Found SIGC2: /usr/lib/arm-linux-gnueabihf/libsigc-2.0.so  
-- Found chown: /usr/bin/chown
-- SvxLink user = svxlink
-- SvxLink group = svxlink
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.4") found components: doxygen missing components: dot
-- Found groff: /usr/bin/groff
-- Found gzip: /usr/bin/gzip
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found DL: /usr/lib/arm-linux-gnueabihf/libdl.a  
-- Found RT: /usr/lib/arm-linux-gnueabihf/librt.a  
-- Found Speex: /usr/lib/arm-linux-gnueabihf/libspeex.so (found version "1.2.1") 
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so (found version "1.3.1") 
-- Found OGG: /usr/lib/arm-linux-gnueabihf/libogg.so (found version "1.3.5") 
-- Checking for module 'alsa'
--   Found alsa, version 1.2.8
-- Checking for module 'popt'
--   Found popt, version 1.19
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so  
-- With Systemd = OFF
-- Looking for HIDIOCGRAWINFO
-- Looking for HIDIOCGRAWINFO - found
-- Found GCrypt: /usr/lib/arm-linux-gnueabihf/libgcrypt.so  
-- Found RtlSdr: /usr/lib/arm-linux-gnueabihf/librtlsdr.so;/usr/lib/arm-linux-gnueabihf/libusb-1.0.so  
-- Found GPIOD: /usr/lib/arm-linux-gnueabihf/libgpiod.so (found version "1.6.3") 
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.5
-- Found Tclsh: /bin/tclsh (found version "8.6") 
-- Found TCL: /usr/lib/arm-linux-gnueabihf/libtcl.so
-- Found CURL: /usr/include/arm-linux-gnueabihf  
-- Package Version = 19.09.1.341.g5c5b05f3
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/src/svxlink/src/build

..................

[  9%] Building CXX object async/cpp/CMakeFiles/asynccpp.dir/AsyncCppDnsLookupWorker.cpp.o
In file included from /opt/src/svxlink/src/build/include/AsyncDnsLookup.h:60,
                 from /opt/src/svxlink/src/async/cpp/AsyncCppDnsLookupWorker.cpp:55:
In member function ‘Async::DnsResourceRecordSRV::Weight Async::DnsResourceRecordSRV::weight() const’,
    inlined from ‘void Async::DnsLookupWorker::addResourceRecord(Async::DnsResourceRecord*)’ at /opt/src/svxlink/src/async/cpp/../core/AsyncDnsLookupWorker.h:327:59,
    inlined from ‘void Async::CppDnsLookupWorker::notificationReceived(Async::FdWatch*)’ at /opt/src/svxlink/src/async/cpp/AsyncCppDnsLookupWorker.cpp:459:28:
/opt/src/svxlink/src/build/include/AsyncDnsResourceRecord.h:657:40: warning: array subscript ‘const Async::DnsResourceRecordSRV[0]’ is partly outside array bounds of ‘unsigned char [36]’ [-Warray-bounds]
  657 |     Weight weight(void) const { return m_weight; }
      |                                        ^~~~~~~~
/opt/src/svxlink/src/async/cpp/AsyncCppDnsLookupWorker.cpp: In member function ‘void Async::CppDnsLookupWorker::notificationReceived(Async::FdWatch*)’:
/opt/src/svxlink/src/async/cpp/AsyncCppDnsLookupWorker.cpp:460:67: note: object of size 36 allocated by ‘operator new’
  460 |               new DnsResourceRecordA(name, ttl, IpAddress(in_addr)));
      |                                                                   ^
In member function ‘Async::DnsResourceRecordSRV::Weight Async::DnsResourceRecordSRV::weight() const’,
    inlined from ‘void Async::DnsLookupWorker::addResourceRecord(Async::DnsResourceRecord*)’ at /opt/src/svxlink/src/async/cpp/../core/AsyncDnsLookupWorker.h:327:59,
[... 51 lines cut ...]
 2164 |  curr= (void*)(((char*)curr) - rv + tmp); rv= tmp; }
      |                ~~~~~~~~~~~~~~^~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2271:48: note: in expansion of macro ‘INCBUF’
 2271 |          while (rvp + sizeof(double) >= rvend) INCBUF;
      |                                                ^~~~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2162:23: note: call to ‘realloc’ here
 2162 | #define INCBUF { tmp= realloc(rv, (rvend-rv) * 2); if (!tmp) error("Out of memory"); \
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2271:48: note: in expansion of macro ‘INCBUF’
 2271 |          while (rvp + sizeof(double) >= rvend) INCBUF;
      |                                                ^~~~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2163:40: warning: pointer ‘rv’ may be used after ‘realloc’ [-Wuse-after-free]
 2163 |  rvend= (rvend-rv) * 2 + tmp; rvp= (rvp-rv) + tmp; \
      |                                    ~~~~^~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2271:48: note: in expansion of macro ‘INCBUF’
 2271 |          while (rvp + sizeof(double) >= rvend) INCBUF;
      |                                                ^~~~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2162:23: note: call to ‘realloc’ here
 2162 | #define INCBUF { tmp= realloc(rv, (rvend-rv) * 2); if (!tmp) error("Out of memory"); \
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/src/svxlink/src/async/audio/fidlib.c:2271:48: note: in expansion of macro ‘INCBUF’
 2271 |          while (rvp + sizeof(double) >= rvend) INCBUF;
      |                                                ^~~~~~

[ 12%] Building CXX object async/audio/CMakeFiles/asyncaudio.dir/AsyncAudioDevice.cpp.o

dg1kwa, 2023-11-16:
on raspbian 12:

make[2]: *** Keine Regel vorhanden, um das Ziel „/usr/lib/arm-linux-gnueabihf/libdl.so“,
  benötigt von „lib/libasynccore.so.1.6.99.26“, zu erstellen.  Schluss.
make[1]: *** [CMakeFiles/Makefile2:1361: async/core/CMakeFiles/asynccore.dir/all] Fehler 2
make: *** [Makefile:156: all] Fehler 2

f5vmr, 2023-11-16:
I recommend bullseye for current applications of svxlink. Until all the
repositories contain new sources this problem will persist.

fabianfrz, 2023-12-22:
I use debian 12 and you can actually install it via the repository. So no need to self compile. But if you need to do so, I guess you could check out the debian src package how they do it.

f5vmr, 2023-12-22:
As of a few days ago the compilation is still not up to date. It does not
include the ReflectorLogic.
I know. I tested it.

f5vmr, 2023-12-22:
I apologise…… It is up to date. Carry on.!

f5vmr, 2023-12-22:
Debian 12

OK I was right - ReflectorLogic is not part of this package….. Can’t beat a compilation from source.

Chris

f5vmr, 2023-12-22:
Not withstanding the lack of ReflectorLogic, I have further discovered that
there is a further issue with using Debian 12 64 bit, is that there are
warnings generated during the compilation method that eventually fails to
generate the library files (.so) for the modules, consequently the whole
compilation is useless. Stick with bullseye for the moment, that is
guaranteed to provide a good build.

ducarpit, 2024-01-02:
Just to report my experience: i have the same problems during the compilation in Debian Bookworm (on a Raspberry) too....
I follow the discussion....

sm0svx, 2024-01-02:
Build warnings for Bookworm fixed in latest master

ducarpit, 2024-01-02:
@sm0svx i can confirm you the fix work! 👍 
I tried the new version just now on my Raspberry with Bookworm and the compilation done successfully!
Thank you so much for your work and support!

closed 2024-01-03 by sm0svx