sm0svx/svxlink #651: GPIO issue orangepi

datatechnician63, 2023-11-04:
I have set  svxlink to use pin 10 as output for PTT and pin 7 as COR, the PTT function is operating normally and will trigger normally with EchoLink connections and normal timed operations however I cannot get the COR to function. Indications confirm COR operation, squelch open and closed, but do not trigger the PTT output. I have tried using a orangepi zero and a orangepi one board and the same is happening, the PTT function works but when the input COR is active the PTT output does not.

f5vmr, 2023-11-04:
I'm unsure of the pin numbering on an orange pi, but the physical pin
numbers on a raspberry pi do not indicate the GPIO pin, for example GPIO 10
is physical pin 19. Have you followed the correct sequencing as described?

datatechnician63, 2023-11-04:
I am using the orangepi gpionumbering gpio10 on pin 26 and gpio7 on pin 12. 

My /sys/rc.local is as follows

# PTT pin
echo “10” > /sys/class/gpio/export
echo “in” > /sys/class/gpio10/direction

# COR pin
echo “7” > /sys/class/gpio/export
echo “out” > /sys/class/gpio7/direction

svxlink.conf GPIO_PIN are set to corresponding gPIO ports

f5vmr, 2023-11-04:
Ok. I see where you are at.
There have been two changes since that former use of GPIO. The first of
those changes was the incorporation of systemd making the GPIO as a
service, with an additional file of gpio.conf. the second change and by far
the most useful has been the addition of GPIOD, that completely changed the
addressing of these pins within the svxlink.conf. The installation of the
latest master version of svxlink, with the pre-install of the gpiod package
in the software list may solve the problem. It will mean you looking
closely at the way you control these pins. If you are still using Jessie,
then persevere with you current version, but if you have upgraded to buster
or bullseye (Debian 10&11), then it will be advantageous to add the
additional software, then look to sudo git pull in the initial download
folder, and sudo make again. If you want to do this, the are posts
describing the gpiod method.
Best wishes.

datatechnician63, 2023-11-11:
I have decided to rebuild svxlink on another orangepi zero h2 running Ubuntu and when I run the cmake line I get the followingCmake error at svxlink/svxlink/CmakeLists.txt:48 (message):Could not find the curl library and development filesSent from my iPhone

f5vmr, 2023-11-11:
Hi,
Then you simply omitted the library prior to the compilation. Apply this line then re-run the sudo cmake line.

sudo apt install -y build-essential g++ make cmake libsigc++-2.0-dev php8.0 nodejs libgsm1-dev libudev-dev libpopt-dev tcl-dev libgpiod-dev gpiod libgcrypt20-dev libspeex-dev libasound2-dev alsa-utils libjsoncpp-dev libopus-dev rtl-sdr libcurl4-openssl-dev libogg-dev librtlsdr-dev groff doxygen graphviz python3-serial

You may find that the OS will report that some of these may already be installed. But if you run this, you shall have no further problems.

Chris

datatechnician63, 2023-11-26:
Doing the dependencies php8.0 was not found

f5vmr, 2023-11-27:
How you install PHP will depend on your current OS - Debian 10 will accept install of php7, Debian 11 will accept install of php8., Debian 12 although untested by me will accept php8.3.

Php has other dependencies that also need installing depending on what else you need. Apache2 add-ons for example. Read up on the dependencies of PHP.

Chris