sm0svx/svxlink #652: ERROR: symbol lookup error: svxlink: undefined symbol

ducarpit, 2023-12-05:
Hi,
in a server with Ubuntu 20.04 LTS and svxlink-server package installed via aptitude package manager i get this error if i run the _svlink_ command:

`svxlink: symbol lookup error: svxlink: undefined symbol: _ZN5Async7AudioIOC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEi`

I also tryed to disinstall and reinstall the packege but the error is the same.
Can anyone help me to understand why?
Thanks a lot
Christian

f5vmr, 2023-12-05:
The apt install is a step or two behind the current svxlink package
normally installed through GitHub. It is version 19.9.1-2, whereas the
latest package is 19.9.2. I installed it myself on Debian 11 a few days
ago, but didn't configure it to run yet.
I'm sorry but the error you described is a bit non-descript, and I've never
encountered it in the past, and I've been using it over 15 years.
I could offer you a script to run the compile version, but it would depend
on what you want to do next.
Chris

f5vmr, 2023-12-05:
> Hi @f5vmr <https://github.com/f5vmr> , yes, this afternoon i founded some
> informations from the Ubuntu package manager and i saw the newest version
> 19.9.2 is only available for Ubuntu 23.10 and not for Ubuntu 22.04 (in that
> case i could upgrade the version from 20.04 to 22.04, but i still wouldn't
> have the newest).
> I don't know why i get this error unfortunatly. Tha main advantage to have
> the repository version is that in the future i can update the package
> without problem, instad with the compiled version is more difficult.
> Do you have a script to compile svxlink simply?
> Thanks a lot,
> Christian

I have, although it is for Debian 11 and suited for the Raspberry pi, is
can be adapted. GitHub.com/f5vmr/svxlink_raspberry. The script may give you
errors in relation to curl and php8.0, Ubuntu 20 would prefer php7.4, but
I'm sure you can review the script to suit. The other thing with the apt
install svxlink will lack ReflectorLogic, whereas the latest svxlink
includes it.
As far as updating goes, it is simpler than you imagine. When in the
install folder sudo git pull will update any changes, and cd src/build
followed by sudo make will perform the updates, followed by sudo ldconfig
to lock it down.
Happy to answer any questions.

Chris

ducarpit, 2023-12-10:
If this can hel someone i SOLVED this problem after some days.
The problem was with the libraries in `/usr/lib/x86_64-linux-gnu`:

- libasyncaudio
- libasynccore
- libasynccpp

Due to some strage reasons, i had in this directory these libraries in a strange version, and i don't know why the `apt install --reinstall` didn't restore them in the right version with the specific symbolic links.
So i backup them from a server with the same SO version, then i deleted them from the server where i got the error and then put the right versions; finally, i re-created the symbolik links by hand to reproduce the same structure from the other server.
And now, the svxlink service works fine! :)
Thank you to all
Christian

closed 2023-12-10 by ducarpit