sm0svx/svxlink #656: LONG_ANNOUNCE without LONG_IDENT! How?

ducarpit, 2023-12-28:
Hi all,
i set up an svxlink server in a Raspberry and in my svxlink.conf configuration file i specified these parameters to manage the IDENT and the ANNOUNCE about my repeater:

```
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=30
#IDENT_ONLY_AFTER_TX=4
#SHORT_ANNOUNCE_ENABLE=1
LONG_ANNOUNCE_ENABLE=1
#SHORT_ANNOUNCE_FILE=
LONG_ANNOUNCE_FILE=/home/pi/sintesi.wav
```

Unfortunatly i have a problem: when the long ident is reproduced also the long announce is reproduce after (cascading).
But if i comment out the LONG_IDENT_INTERVAL parameter also the long announce isn't reproduce.
Is there a way to reproduce only the long announce file at every LONG_IDENT_INTERVAL time without the long ident first?
Thanks you so much in advance

Christian

f5vmr, 2023-12-29:
Dependent on where you have the .tcl files /usr/share/svxlink/events.d/ or
one more folder deep ....local/ - sudo nano Logic.tcl. you will see the
definition of the announcements. I've not done it, but you can set voice as
0 and CW as 1....
Will this solve the problem?

Chris g4nab

ducarpit, 2023-12-29:
> Dependent on where you have the .tcl files /usr/share/svxlink/events.d/ or one more folder deep ....local/ - sudo nano Logic.tcl. you will see the definition of the announcements. I've not done it, but you can set voice as 0 and CW as 1.... Will this solve the problem? Chris g4nab
> […](#)

Hi,
do you think is a good idea to modify the .tcl file directly? Yes, they are in that folder (/usr/share/svxlink/events.d/).
What are "voice" and "CW" parameters? What they control?
Thank you in advance
Christian

f5vmr, 2023-12-29:
> … <x-msg://5/#>

Hi Christian,

In my build script, I ensure that a new folder ‘local’ is created inside /usr/share/svxlink/events.d/ into which I force the copy of all .tcl files. This allows the original files to remain intact in case of problems. Svxlink will always use the ‘local' files
from that point on. So I in fact modify the Logic.tcl file in the local folder. Normally the voice files are activated by default, and the CW aspects are set to 0. These just enable the rest of the logic to activate this parts when required.

You can do the same as follows. 
cd /usr/share/svxlink/events.d
sudo mkdir local
sudo cp *.tcl local/ 
Then
cd local 
To edit the features required.

Chris

ducarpit, 2023-12-31:
Hi @f5vmr ,
at the end i solved my problem.
By reading the Logic.tcl you suggested me some days ago, i founded some parameters to add in the svxlink.conf configuration file to exclude the ident reading before the LONG_ANNOUNCE.
This is the code inside the Logic.tcl:

```
proc send_long_ident {hour minute} {
  global mycall;
  global loaded_modules;
  global active_module;
  variable CFG_TYPE;
  variable long_announce_file
  variable long_announce_enable
  variable long_voice_id_enable
  variable long_cw_id_enable

  # Play the voice ID if enabled
  if {$long_voice_id_enable} {
    puts "Playing Long voice ID"
    spellWord $mycall;
    if {$CFG_TYPE == "Repeater"} {
      playMsg "Core" "repeater";
    }
    playSilence 500;
    playMsg "Core" "the_time_is";
    playSilence 100;
    playTime $hour $minute;
    playSilence 500;

    # Call the "status_report" function in all modules if no module is active
    if {$active_module == ""} {
      foreach module [split $loaded_modules " "] {
        set func "::";
        append func $module "::status_report";
        if {"[info procs $func]" ne ""} {
          $func;
        }
      }
    }

    playSilence 500;
  }

  # Play announcement file if enabled
  if {$long_announce_enable} {
    puts "Playing long announce"
    if [file exist "$long_announce_file"] {
      playFile "$long_announce_file"
      playSilence 500
    }
  }

  # Play CW id if enabled
  if {$long_cw_id_enable} {
    puts "Playing long CW ID"
    if {$CFG_TYPE == "Repeater"} {
      set call "$mycall/R"
      CW::play $call
    } else {
      CW::play $mycall
    }
    playSilence 100
  }
}
```

As you can see, there is a variable _$long_voice_id_enable_ and its default value (from code) is 1 (enabled). By reading the svxlink manual page, i founded some parameters to add in the svxlink.conf configuration file to manage these behaviors:

```
SHORT_VOICE_ID_ENABLE=1
LONG_VOICE_ID_ENABLE=0
```

By set the LONG_VOICE_ID_ENABLE at 0 i can exclude the ident reading during the long announce.
Thank you so much for your suggestion, i hope this recap can help other users with the same problem.
Christian IU6DJR

closed 2023-12-31 by ducarpit