sm0svx/svxlink #658: Informations about svxlink compilation and services

ducarpit, 2024-01-01:
Hi all,
i need two informations about the svxlink building:

1) I compiled svxlink by following the informations in the wiki project, so at the end i compiled and installed it and now i can run the service correctly. In the past i used to install svxlink and svxreflector by using the standard package include in the Ubuntu and Raspbian standard repository and with this one i had all the files to control the services with systemd (the .service files in /lib/systemd/system); there were also the files to rotate the log (in /etc/logrotate.d) and the files with the environment viriables (in /etc/default)...in my case, instead, i don't have these files and i don't know if this normal or if i wrong something during the svxlink building.

2) If i compile svxlink in a server, is there a way to run the executable files in another server with the same architecture and operating system with the same libraries installed? I tried to copy the files /usr/bin/svxlink and /usr/bin/svxreflector to another server but i get the error "unedefined symbol" unfortunatly when i run the services.... 

Thank you so much in advance
Christian IU6DJR

f5vmr, 2024-01-01:
Hi again Christian.
The source package at https://github.com/sm0svx/svxlink.git being the
latest version contains everything for all the packages, simplex, repeater,
remotetrx and svxreflector. Everything is compiled at the same time. How it
is run, is dependant on which service is activated. If you activate it
using svxlink.service, then you use svxlink.conf of course. If you have set
it up in a PC for a reflector then set svxreflector.service using
svxreflector.conf. There have been various reports of failures using Debian
12, probably due to lack of repositories for some of the required
softwares. Stick with Debian 11 or a version of Ubuntu 20 that is proven to
be compatible. There is a list of software to be installed prior to
compiling SVXLink so check you have not missed one of these. My server
running Debian 11 works very well, and my raspberry svxlink nodes are all
running on bullseye.
As Debian 11 appears to be long term, I see little advantage in upgrading.

If you are copying .conf files, you will find no change. It is the c++ and
compiled library files you need.

If you go to your original installation file svxlink and sudo git -v to
check your source.
sudo git remote -v —— this will confirm you have a branch installed. It
should be origin/master.
sudo git branch -r —— this will show you the branches available.
sudo git checkout .... Then the correct branch will update your current
repository.
Then cd src/build followed by sudo make to recompile.
sudo make install followed by sudo ldconfig will make it ready for use
again.

Chris

ducarpit, 2024-01-01:
Hi @f5vmr ,
i am using Ubuntu 22.04 LTS and i didn't get any error during the compilation.
The branch was correct. I found this issue: https://github.com/sm0svx/svxlink/issues/242
Maybe i need to specify the -DWITH_SYSTEMD=ON on my cmake command?
Thank you so much
Christian IU6DJR

f5vmr, 2024-01-01:
Yes, you can use the same install files, just start with sudo make clean in
svxlink/src/build, before running the sudo cmake with the added
specification you have found, for the systemd.
Chris

ducarpit, 2024-01-02:
Hi @f5vmr  ,
yes, in this way the systemd configuration file are been generated and then they are been installed in the correct directory during the installation process 👍 
I saw that there are also the configuration files for the log rotation but they are not copied in the right directory (/etc/logorotate.d)....i can copy them by hand, do you know if are there some variables to pass tot the cmake command to enable this copy?

Do you have any idea for my second question on the top instead?
Thank you so much in advance
Christian

f5vmr, 2024-01-02:
I have never used log rotate, I made my own script. There is an issue with
the build of svxlink. If you look at /etc/default/svxlink you will see that
the name of the log is not svxlink.log but svxlink. I run a script on my
builds that changes this profile to a more acceptable standard. I also then
add a crontab that runs a bash script at 23:59 daily, that copies the log
to a text file, and resets the log to zero byes. The bash script also
removes any log.txt older than seven days. So it keeps everything tidy.
You can view all my script files at github.com/f5vmr/svxlink_raspberry.
Chris

ducarpit, 2024-01-03:
Ok, thank you @f5vmr ...
I report my conclusions if they can help other users in the future:

1) About the first question, i needed to add the parameter `-DWITH_SYSTEMD=ON` on the cmake command to generate the systemd configuration files that are installed then during the install process

2) About the second question, i compiled svxlink in a server by using the `-DCMAKE_INSTALL_PREFIX=/opt/[MY_DIRECTORY]` so all the files were put in that direcory; then, in the other server i installed first the libraries and packages needed for the svxlink services (without installing the dev libraries) and finally i copied the compiled files and folder from the first server in the right location in the new server (pay attention to the paths in the configuration files).
It is not very simple and i don't knwo if this is the best way to solve my need, but at the moment it's the only way i found....

f5vmr, 2024-01-03:
Hi Christian,
If it helps, this is the standard compile line.
sudo cmake -DUSE_QT=OFF -DCMAKE_INSTALL_PREFIX=/usr
-DSYSCONF_INSTALL_DIR=/etc \ -DLOCAL_STATE_DIR=/var -DWITH_SYSTEMD=ON ..
Chris

closed 2024-01-03 by ducarpit