sm0svx/svxlink #659: svxreflector: status page not working

marcels1990, 2024-01-12:
hello,

i have 2 working svxreflectors running on a virtual machine on debian without problems. I was installing a svxreflector for another ham group on a raspberry and they wanted to access the status page (http://IP_REFLECTOR:8080/status). But the raspberry the page is not working, using netstat i only see the port 5300 open. On my other reflector the page is working without problems.
I tried to change the port number on the config file of the reflector but without success. Can you give me some advice please?

thank you very much

f5vmr, 2024-01-12:
I use 8080 for another machine on my network.  I also run svxportal in the
web server, I have forced the port to 8181, and port-forwarded this to the
external address. I clearly see the json content of all the connected
stations. I also run the reflector and the portal on a VM, and I use a
external address to access both the portal and the web server.

marcels1990, 2024-01-12:
Ok thank you. Can you give me some advice on the raspberry? I have do not have an idea why the port does not come up.Thank you

f5vmr, 2024-01-12:
All I can suggest that on the router to which you connect the raspberry,
that you port-forward 8080 from the internal to the external address. I do
a lot of port redirection on my devices, as I run a dmr dashboard too, as
well as my own web page. Quite frankly, (franchement) I do not use the 8080
capability because of the user of svxportal, even though I have redirected
to 8181 instead. (See svxportal-uk.ddns.net:81 as an example). For more
information on this, go to github.com/sa2blv .
73
Chris

marcels1990, 2024-11-08:
Solved after reinstall, thank you

closed 2024-11-08 by marcels1990