sm0svx/svxlink #664: Remove the roger beeps/CW digits on repeater tail with Echolink node!

ducarpit, 2024-03-01:
Hi all,
some days ago i asked about this problema in the community but unfortunatly no solution was founded.
I have a "problem" with my Svxlink installation where is configured an Echolink node: when a user is connected to the Echolink node, a sound similar a roger beep or a CW is reproduced at the end of the repeater tail (similar to a three or four beep digits) and it is reproduced by the TX radio, so all the ham operators that are listening can hear this sound, instead the Echolink users don't hear it.
You can hear and see it by the video in attachment:

https://github.com/sm0svx/svxlink/assets/17808926/956b3ca7-df2c-4dc0-83c7-174f16abb6b6

I tried to look for a specific function or a specific instruction in the TCL file (mostly in the Echolink.tcl and Logic.tcl) but i didn't find anything about this.
Can anyone help me to understand how to disable this annoying sound or to understand if this is a bug?
Thanks a lot in advance
Christian

f5vmr, 2024-03-01:
I replied previously for the ModuleEchoLink.tcl aspect but in Logic.tcl this is the problem you may have.

The procedure depends on wether you have set up the RK token or TX token at the head of those aspects in svxlink.conf.
Look for this procedure.

proc send_rgr_sound {} {
  variable sql_rx_id

  if {$sql_rx_id != "?"} {
    # 200 CPM, 1000 Hz, -10 dBFS
    CW::play $sql_rx_id 200 1000 -10
    set sql_rx_id "?"
  } else {
    playTone 440 500 100
  }
  playSilence 100
}

In the case where the token is set then CW will be sent according to the value of the token. If “K” then a “K” will be sent in CW.
You should then considering commenting out the token line. However then the single tone will be sent. So comment out this also,
In my repeater I comment out the token completely, but I change the playTone line to CW::play “K” 95 650 -10 to send a “K” every time.

Chris G4NAB

ducarpit, 2024-03-05:
Hi @f5vmr and thank you for your support!
I tried in the way you described and now the beep(s) seem disappeared! :)
Probably this was the exact point where i need to comment out some instructions.
Thank you so much
Christian de IU6DJR

closed 2024-03-12 by ducarpit