sm0svx/svxlink #665: REQ: svxreflector implementaion for SIGHUP (re-read config without restart)

audric, 2024-03-15:
I wish svxreflector daemon implemented the SIGHUP signal in order to re-read the config file without having to kill the process and restarting it.

This would be useful to enforce config changes (like new users or user deletions) without affecting the present connections.

Open to other suggestions that achieve the same goal.

f5vmr, 2024-03-15:
It find it takes only seconds for the implementation of a restart, that
barely makes a flicker on the portal dashboard. I see no real issues, it's
not a professional communications system however you look at it, as good as
it is.
Chris

sm0svx, 2025-09-28:
As @f5vmr commented, a reflector restart is normally hardly noticeable even when in the middle of a QSO.
As for user additions/deletions, you should not use the AUTH_KEY authentication anymore but certificates.

A complete re-read of the config would be a lot of work to implement so it will not happen. A couple of variables are settable at run-time using the COMMAND_PTY (GLOBAL/SQL_TIMEOUT_BLOCKTIME and GLOBAL/SQL_TIMEOUT). And if you really want to, you can also add entries to the USERS and PASSWORDS section at run-time. They will not be stored in the config file though so you will have to do that manually.

sm0svx, 2025-09-28:
On the reflector side it's not a problem. The latest reflector software will support older clients just fine so start by upgrading those old reflectors. Nothing will change really until some node connect using the V3 protocol.

f5vmr, 2025-09-28:
> On the Swedish reflector the oldest V2 node is running SvxLink 1.7.99.52
> and all reflectors are obviously running the latest development code from
> git master.

Ok, I'll check it out.
Thanks
Chris