sm0svx/svxlink #670: SVXLINK holds transmit open for a long time

phlpittsny, 2024-04-10:
I have svxlink (version 19.09.1-3) working, to a point. But, when I elicit a response, it does not stop transmitting (i.e., it holds the PTT key down) for what seems like a long time. Need help, please.
System: Raspberry Pi 3B connected to a Baofeng UV-82 (primary radio) ... see below.
 Secondary Baofeng UV-82 transmitting to the primary radio on 438.875MHz
 Both radios set for CTCSS of 110.9Hz (Transmit AND Receive)
I start svxlink from the console of the RPi.
 On the secondary radio I press PTT and then *# and the primary radio starts to transmit the
 station identification, and on the console svxlink announces "Tx1: Turning the transmitter ON".
At the end of the station id, svxlink announces "Tx1: Turnng the transmitter OFF" ... but does
 NOT stop transmitting for about a full minute.
Is this correct operation, or should the transmitter shut off sooner?

System details:
/etc/rpi-issue: on the RPi
 Raspberry Pi reference 2024-03-12
 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b9b581e0a62afb7200d4b0b72ec2c43485fd49cc, stage4
from /etc/svxlink/svxlink.conf
 Configured for PTT via GPIO18
[Tx1]
 TYPE=Local
 AUDIO_DEV=alsa:plughw:1
 AUDIO_CHANNEL=0
 #AUDIO_DEV_KEEP_OPEN=0
 PTT_TYPE=GPIO
 PTT_PORT=GPIO
 PTT_PIN=gpio18
 #HID_DEVICE=/dev/hidraw3
 #HID_PTT_PIN=GPIO3
 #SERIAL_SET_PINS=DTR!RTS
 #GPIO_PATH=/sys/class/gpio
 PTT_HANGTIME=1000
 TIMEOUT=300
 TX_DELAY=500
 CTCSS_FQ=110.9
 #CTCSS_LEVEL=9
 PREEMPHASIS=0
 DTMF_TONE_LENGTH=100
 DTMF_TONE_SPACING=50
 DTMF_DIGIT_PWR=-15
 #MASTER_GAIN=0.0
 #OB_AFSK_ENABLE=0
 #OB_AFSK_VOICE_GAIN=-6
 #OB_AFSK_LEVEL=-12
 #OB_AFSK_TX_DELAY=100
 #IB_AFSK_ENABLE=0
 #IB_AFSK_LEVEL=-6
 #IB_AFSK_TX_DELAY=100

The RPi is connected to the primary radio (BAOFENG UV-82) via an Easy Digi Kit
 board with the following connections
Rpi to Digi Kit
 Pin9 (GND) to Digi Kit Dig Gnd
 Pin12 (GPIO18) to Digi Kit DTR
 USB Sound Card MIC to Digi Kit "Audio to PC"
 USB Sound Card SPK to Digi Kit "Audio from PC"
Digi Kit to Radio
 Digi Kit Audio GND + MIC GND + PTT GND to Radio GND (2.5mm plug Shield (S of TRS))
 Digi Kit Audio Out to Radio SPK+ (2.5mm plug Tip (T of TRS))
 Digi Kit MIC IN to Radio Mic+ (3.5mm plug Ring (R of TRS))
 Digi Kit PTT HI To Radio PTT (3.5mm plug Shield (S of TRS))
Thanks in advance for any help you can give.

f5vmr, 2024-04-10:
Hello,

First note is that you have PTT_HANGTIME=1000 which means it stays on for 1000 Milliseconds, suggest you reduce it to 10 or even 0.

Are you UK based?

Chris G4NAB

PS. New versions of the package is now 24.01.8

phlpittsny, 2024-04-10:
Thanks... I'll try that. I'm in Rochester  NY ... USA.

I have another problem if you can help... svxlink only responds to some
(not all) DTMF digits. Any ideas?

f5vmr, 2024-04-10:
Certainly - if you ensure the DEEMPHASIS=0 in the [Rx1] that may help.

Chris

phlpittsny, 2024-04-10:
I'll try that, too. Thanks...that helps a lot.