sm0svx/svxlink #675: Can no longer install SVxLink

DL7TA, 2024-07-22:
I've been trying to install SVxLink since yesterday.
Cloning works fine, but the make command stops after 8% with the following error message:
In file included from /home/pi/svxlink/src/build/include/AsyncTcpClientBase.h:52                                                                                                                                                             ,
                 from /home/pi/svxlink/src/build/include/AsyncTcpClient.h:58,
                 from /home/pi/svxlink/src/build/include/AsyncTcpPrioClient.h:50                                                                                                                                                             ,
                 from /home/pi/svxlink/src/svxlink/svxlink/ReflectorV2Logic.h:50                                                                                                                                                             ,
                 from /home/pi/svxlink/src/svxlink/svxlink/ReflectorV2Logic.cpp:                                                                                                                                                             61:
/home/pi/svxlink/src/build/include/AsyncTcpConnection.h:42:10: fatal error: open                                                                                                                                                             ssl/bio.h: Datei oder Verzeichnis nicht gefunden
   42 | #include <openssl/bio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [svxlink/svxlink/CMakeFiles/ReflectorV2Logic.dir/build.make:76: svx                                                                                                                                                             link/svxlink/CMakeFiles/ReflectorV2Logic.dir/ReflectorV2Logic.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:2704: svxlink/svxlink/CMakeFiles/ReflectorV2L                                                                                                                                                             ogic.dir/all] Fehler 2
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....

The error is reproducible.
Has the installation method changed since updating to the latest version?

vy 73!
Dan DL7TA

sp2ong, 2024-07-22:
You must install library:

sudo apt install openssl-dev 

and run again cmake

sm0svx, 2024-07-22:
There is a new depencency on the OpenSSL library. The package can be named different things on different platforms. On my Raspberry I installed libssl-dev.

closed 2024-07-22 by sm0svx