sm0svx/svxlink #677: GPIO cannot be activated

DL7TA, 2024-08-02:
Hello OM.
I have been successfully operating several repeaters with Svxlink for years and have written a small manual for the installation and configuration. Up until now, the installation worked without any problems, but unfortunately it is currently stuck.

I configure my GPIO pins as follows in /etc/svxlink/gpio.conf: 
###############################################################################
#
# Configuration file for the SvxLink server GPIO Pins
#
###############################################################################

# GPIO system pin path
# RPi/odroid/nanopi/pine64 = /sys/class/gpio, orangpi = /sys/class/gpio_sw
GPIO_PATH=/sys/class/gpio

# Space separated list of GPIO pins that point IN and have an
# Active HIGH state (3.3v = ON, 0v = OFF)
GPIO_IN_HIGH="gpio26"

# Space separated list of GPIO pins that point IN and have an
# Active LOW state (0v = ON, 3.3v = OFF)
GPIO_IN_LOW=""

# Space separated list of GPIO pins that point OUT and have an
# Active HIGH state (3.3v = ON, 0v = OFF)
GPIO_OUT_HIGH="gpio13"

# Space separated list of GPIO pins that point OUT and have an
# Active LOW state (0v = ON, 3.3v = OFF)
GPIO_OUT_LOW=""

# User that should own the GPIO device files
GPIO_USER="svxlink"

# Group for the GPIO device files
GPIO_GROUP="svxlink"

# File access mode for the GPIO device files
GPIO_MODE="0664"

I activate the GPIO with sudo systemctl enable svxlink_gpio_setup and then start Svxlink.

This starts with the error message:
"*** ERROR: Could not open GPIO device /sys/class/gpio/gpio26/value specified in Rx1/GPIO_SQL_PIN: No such file or directory"

Svxlink did not create any folders for the individual GPIOs under /sys/class/gpio. 
This had always worked without problems up until now.

Has something changed in the current version of Svxlink so that I now have to take a different approach?

vy73!
Dan

f5vmr, 2024-08-02:
The new version of svxlink use gpiod and not gpio.conf, which should remain
but be unconfigured. Both gpiod and lib2gpiod-dev should be installed with
apt. The svxlink.conf in Rx1 and Tx1 should indicate the presence of
gpiochip0 or similar. The man file should contain the relevant changes.
Come back if you need more detailed help.
Chris

DL7TA, 2024-08-04:
Hello Chris,
thanks for your answer.
Unfortunately the setting with SQL_DET=GPIOD doesn't work.
Svxlink is up to date, but the error message still appears:
ERROR: Unknown squelch type specified in config variable Rx1/SQL_DET. Legal squelch types are: "COMBINE" "CTCSS" "EVDEV" "GPIO" "HIDRAW" "OPEN" "PTY" "SERIAL" "SIGLEV" "VOX"
The docs here on Github say that the following is possible:
.B SQL_DET
Specify the type of squelch detector to use. Possible values ​​are: VOX, CTCSS,
SERIAL, EVDEV, SIGLEV, PTY, GPIO, GPIOD, HIDRAW or COMBINE.
But unfortunately it doesn't work.
Do you have any other ideas?
vy 73!
Dan

f5vmr, 2024-08-04:
Two things come to mind at this point. Your version of svxlink-server is
seemingly not up to date as it should be as it doesn't contain the gpiod
squelch option, and gpiod may not be installed by apt.
I'm working with version 24-02.1 on Debian 12 bookworm. I have perfected
(almost) the complete script that will solve your issues. If you wish to
try this, then email me direct. There is no compilation for this build, and
can be operational in probably less than 20 minutes.

yoch2015, 2024-08-30:
Hello,
I faced the same problem and I solved it so I'll share it.
It seems that the new OS has changed the way GPIO is accessed.

First, generate a path for gpio26.

$ cat /sys/kernel/debug/gpio | grep GPI26
gpio-538 (GPI26               )

I found out it is gpio-538, so I will export it.

$ echo 538 > /sys/class/gpio/export
$ cat /sys/kernel/debug/gpio | grep GPI26
gpio-538 (GPI26               |sysfs               ) in  hi

Generated in/hi.

Next, PTT (GPIO13) is generated in the same way.
The ptt is also generated with in/hi, so it changes direction.

$ echo out > /sys/class/gpio/gpio525/direction
$ cat /sys/kernel/debug/gpio | grep GPI13
gpio-525 (GPI13               |sysfs               ) out lo

Make svxlink.conf as follows.
If you want the opposite behavior, remove or add the [!].

svxlink.conf

SQL_DET=GPIO
GPIO_PATH=/sys/class/gpio
GPIO_SQL_PIN=!gpio538

PTT_TYPE=GPIO
PTT_PIN=gpio525
GPIO_PATH=/sys/class/gpio

73!

f5vmr, 2024-08-30:
I did overcome this problem with W6NAS repeater which has this
configuration. Gpiod works with it as well. I'll go back to the sysop to
recall what we did.

closed 2024-12-30 by sm0svx

petrkr, 2025-04-21:
GPIO SysFS has been deprecated almost 15 years ago.

https://www.kernel.org/doc/Documentation/gpio/sysfs.txt
https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/

you should already use gpiod, but I lack documentation about it in svxlink

sm0svx, 2025-04-21:
You find info about how to set up GPIOD in the svxlink.conf manual page. Log in to your system and type "man svxlink.conf" then type /GPIOD to search for info. If you does not find anything you need to upgrade SvxLink to a newer version.

sm0svx, 2025-04-21:
Never use the web page as a reference. The correct documentation is always in the manual page on your system.