sm0svx/svxlink #678: Echolink Module won't start

YO3LNK, 2024-09-04:
Hello. I'm making a ham network and i have a svxreflector.conf file the svxlink.conf main file and moduleecholink.conf file. This is on the server side. I'm trying to connect with echolink but the script doesn't even start.

svxreflector.conf:

```
[GLOBAL]
TIMESTAMP_FORMAT="%c"

[ReflectorLogic]
TYPE=Reflector
PORT=5300
HOST=0.0.0.0
CALLSIGN=YO3LNK-L
AUTHORIZED_IPS=*
MAX_CONNECTIONS=100
AUTH_KEY=mysecurekey

[USERS]
YO3LNK-L=SERVER

[PASSWORDS]
SERVER=mysecurekey
```

svxlink.conf:

```
[GLOBAL]
LOGICS=ReflectorLogic
MODULES=ModuleEchoLink
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"

[ReflectorLogic]
TYPE=Reflector
PORT=5300
HOST=127.0.0.1
CALLSIGN=YO3LNK-L
AUTHORIZED_IPS=*         
MAX_CONNECTIONS=100      
AUTH_KEY=mysecurekey
```

svxlink.d/ModuleEchoLink.conf:

```
[ModuleEchoLink]
NAME=EchoLink
ID=290272
TIMEOUT=60
#ALLOW_IP=192.168.1.0/24
#DROP_INCOMING=^()$
#REJECT_INCOMING=^()$
#ACCEPT_INCOMING=^(.*)$
#REJECT_OUTGOING=^()$
#ACCEPT_OUTGOING=^(.*)$
#REJECT_CONF=0
#CHECK_NR_CONNECTS=2,300,120
SERVERS=europe.echolink.org
CALLSIGN=YO3LNK-L
PASSWORD=mypass
SYSOPNAME=Stefan
LOCATION=[Svx] Fq, Bucharest
#PROXY_SERVER=the.proxy.server
#PROXY_PORT=8100
#PROXY_PASSWORD=PUBLIC
#BIND_ADDR=10.20.30.40
# MAX_QSOS=10
MAX_CONNECTIONS=11
LINK_IDLE_TIMEOUT=300
#AUTOCON_ECHOLINK_ID=9999
#AUTOCON_TIME=1200
#USE_GSM_ONLY=1
DEFAULT_LANG=en_US
#COMMAND_PTY=/dev/shm/echolink_ctrl
#REMOTE_RGR_SOUND=0
```

Please help me

f5vmr, 2024-09-04:
Hi Stefan, To enable ModuleEchoLink, you must have an SimplexLogic or a
RepeaterLogic running too. It will not function on its own. I can help you,
but please be patient as I am preparing a meal at the moment. I will share
you a copy of how it should be configured.

YO3LNK, 2024-09-04:
Thank you so much for the quick reply. I'm ok thanks. I solved it and now it's no problem. Thanks a lot!

closed 2024-09-04 by YO3LNK