sm0svx/svxlink #679: Dynamic change of LADSPA plugin parameters.

ur3qjw, 2024-09-05:
Hi Tobias,

Is it possible to somehow dynamically change the values ​​of LADSPA plugin parameters?
In process of work. Without restarting SvxLink.
This can be used, for example, to adjust values ​​depending on the RSSI level of reception. Possibly, at certain RSSI threshold values.

Is it possible to add support for LADSPA plugins to the transmitter in the future?

Best regards,
Volodymyr

sm0svx, 2024-09-05:
Hi Volodymyr,

It would be possible to add dynamic changing of LADSPA parameters. The most straight forward way to implement that would be to make it possible to set the LADSPA_PLUGINS configuration parameter at runtime. Not very practical maybe but pretty easy to implement. Then that configuration variable would be possible to set via the COMMAND_PTY and from TCL using the setConfigValue function. A more practical interface for changing parameters at runtime would require some new mechanism and would involve a lot more work.

Adding LADSPA to TX would also probably be easy.

If you are looking for a way to find good static parameter settings one way is to record representative audio in SvxLink (e.g. using QSO recorder) and then load the LADSPA plugin and audio file into a tool like audacity where you more easily can tweak LADSPA plugin settings.

sm0svx, 2024-09-07:
LADSPA transmitted audio shaping added in commit b66719b8282df1e1300f583c389e25116730a59f

ur3qjw, 2024-09-08:
Hi Tobias,

Thank you very much for supporting the plugins for the transmitter!

> ... Then that configuration variable would be possible to set via the COMMAND_PTY and from TCL using the setConfigValue function. ...

Is it possible to pass parameters from a file or script?
If use several plugins, there can be many parameters changed.
It is difficult to take them all into account in a TCL file.

I'm trying to try using a noise reduction plugin for the EchoLink gateway and SvxReflector.
For this I used 2 simplex logics.
But, even without the use of modules and plugins. SvxLink loads the CPU to 43%.
More details [here: #326](https://github.com/sm0svx/svxlink/issues/326#issuecomment-2330267641)
I think this is what Waldek was talking about. When the load reached 100% when using plugins together with Alsa_Loopback.

Best regards,
Volodymyr

sm0svx, 2024-09-09:
I came up with a solution for setting single LADSPA configuration parameters at runtime. With commit ede2ef44ca35f34449461bbaa45cabdcd370d317 you can specify LADSPA parameters in a separate configuration section, like this:
```
[Rx1]
...
LADSPA_PLUGINS=hpf:500,@Rx1_Compressor

[Rx1_Compressor]
LABEL=tap_dynamics_m
Attack=4
Release=500
Offset_Gain=15
Makeup_Gain=15
Function=13
```
With this setup you can use the COMMAND_PTY or setConfigValue from TCL to set a specific LADSPA parameter at runtime.

ur3qjw, 2024-10-01:
Thank you very much Tobias!

closed 2024-10-01 by ur3qjw