sm0svx/svxlink #680: SVXLink v1.8.0 Seg.fault on AlpineLinux

ur3qjw, 2024-10-01:
Hi All,

SvxLink, crashes when using the ModuleEchoLink and ReflectorLogic.
With the ModuleFrn, it starts normally.
Here are the build logs:
[build-edge-armhf](https://build.alpinelinux.org/buildlogs/build-edge-armhf/community/svxlink/svxlink-24.02-r2.log)
[build-edge-armv7](https://build.alpinelinux.org/buildlogs/build-edge-armv7/community/svxlink/svxlink-24.02-r2.log)
[build-edge-x86_64](https://build.alpinelinux.org/buildlogs/build-edge-x86_64/community/svxlink/svxlink-24.02-r2.log)

Maybe someone can tell how to solve this problem?

Debugger information:
`SvxLink v1.8.0@24.02 Copyright (C) 2003-2023 Tobias Blomberg / SM0SVX
SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.
Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz
Starting logic: SimplexLogic
	Found plugin: /usr/lib/svxlink/SimplexLogic.so
SimplexLogic: Loading RX "NONE"
SimplexLogic: Loading TX "NONE"
SimplexLogic: Loading module "ModuleEchoLink"
	Found /usr/lib/svxlink/ModuleEchoLink.so
	Module EchoLink v1.6.0@24.02 starting...
Program received signal SIGSEGV, Segmentation fault.
0xb6fb7e90 in memcpy () from /lib/ld-musl-armhf.so.1
(gdb) bt full
#0  0xb6fb7e90 in memcpy () from /lib/ld-musl-armhf.so.1
No symbol table info available.
#1  0xb676bb54 in bool Async::Config::getValue<std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) const () from /usr/lib/svxlink/ModuleEchoLink.so
No symbol table info available.
#2  0xb6766ac4 in ModuleEchoLink::initialize() () from /usr/lib/svxlink/ModuleEchoLink.so
No symbol table info available.
#3  0x0045b5c0 in Logic::loadModule(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    ()
No symbol table info available.
#4  0x0045c600 in Logic::loadModules() ()
No symbol table info available.
#5  0x0046004c in Logic::initialize(Async::Config&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
No symbol table info available.
#6  0xb6803c08 in SimplexLogic::initialize(Async::Config&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/svxlink/SimplexLogic.so
No symbol table info available.
#7  0x004485e4 in ?? ()
No symbol table info available.
#8  0x004457dc in main ()
No symbol table info available.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz
Starting logic: SimplexLogic
	Found plugin: /usr/lib/svxlink/SimplexLogic.so
SimplexLogic: Loading RX "NONE"
SimplexLogic: Loading TX "NONE"
SimplexLogic: Event handler script successfully loaded.
Starting logic: ReflectorLogic
	Found plugin: /usr/lib/svxlink/ReflectorLogic.so
Program received signal SIGSEGV, Segmentation fault.
0xb6fb7e90 in memcpy () from /lib/ld-musl-armhf.so.1
(gdb) bt full
#0  0xb6fb7e90 in memcpy () from /lib/ld-musl-armhf.so.1
No symbol table info available.
#1  0xb67312a4 in bool Async::Config::getValue<std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, bool) const () from /usr/lib/svxlink/ReflectorLogic.so
No symbol table info available.
#2  0xb6722e60 in ReflectorLogic::initialize(Async::Config&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/svxlink/ReflectorLogic.so
No symbol table info available.
#3  0x004485e4 in ?? ()
No symbol table info available.
#4  0x004457dc in main ()
No symbol table info available.`

Best regards,
Volodymyr

sm0svx, 2025-02-08:
Does latest master branch resolve this problem? It looks like this could be caused by one of the problems that I recently fixed.

sm0svx, 2025-02-11:
Fixed according to: https://groups.io/g/svxlink/topic/105368134

closed 2025-02-11 by sm0svx