sm0svx/svxlink #683: svxlink 1.8.99.9 not starting only display version

sp2ong, 2024-10-28:
Currently I have svxlink 1.8.99.8 and it works correctly

I have an interesting phenomenon with version 1.8.99.9 of the source, the new source code was downloaded in its entirety, the compilation on DEBIAN 11 was done correctly and after installation and starting svxlink does not work.

I decided to manually start this new version of go and I see that trying to start the new svxlink ends with showing only the svxlink version
and returning to the command line.

./svxlink --logfile=/var/log/svxlink --config=/etc/svxlink/svxlink.conf --runasuser=svxlink
1**.8.99.9@24.02-46-g7a883880**

./svxlink  --config=/etc/svxlink/svxlink.conf --runasuser=svxlink
**1.8.99.9@24.02-46-g7a883880**

I went back to svxlink 1.8.99.8 and everything works correctly

sp2ong, 2024-10-28:
I'm not a programmer, but while looking for a solution to a problem I had, I found this comment:

_The global variable are by default set to zero, and local variable are using current value from that memory location and it s good practice to initialize variable to zero._ 

For some reason when I compile a new svxlink it looks like the print_version variable is set to something other than 0 which causes the svxlink execution to end with the program version being printed and the condition is exit 0 so the program exits.
Moving the print_version ini to the place where the variables are declared causes the entire code to work correctly.
Maybe this is a special case on my Debian 11?

sp2ong, 2024-10-28:
I check on Debian 12 ArmBian Ornage Pi Zero 3  svxlink v1.8.99.9 has the same problem described, i.e. running svxlink causes the version to be printed and stopped

Debian 11 on Armbian Orange Pi Zero v1 works without problem described
Debian 11 on RaspBian Raspberry PI 3  works without problem described

sm0svx, 2024-10-28:
Sorry, I was a bit sloppy with that one (adding --version switch). It should work now.

closed 2024-10-28 by sm0svx