sm0svx/svxlink #684: Svxlink exporting audio line levels as system variable

cbspidi, 2024-11-02:
Hi Guys, 

Can I request a feature, please?

Working with svxreflector for some time, I find it challenging to find the actual audio line levels.(repeater works in server (CLI) mode only.
It would be super useful to know what the svxlink audio line levels are,  reported in dB value. 
It would be very nice to have it exported into a system variable called `$svx_tx_line_level`  and `$svx_rx_line_level` respectively. 

This could be partially implemented already, a form of alert has already been produced in the logs for peak detection.

Automated actions can be written based on specific values for those variables.

Please let me know what you think.

f5vmr, 2024-11-02:
Hi,
I am unsure what you mean by audio line levels.
In my systems and scripts I previously used alsamixer, but I now implement
amixer that has a read and write facility, and reads and writes specific to
specific numid variables using sget and cget. You can get the various
arguements from the options using amixer.

In my systems, I used .php in the dashboard to implement the changes by
giving limited and specific access of the webserver to sudo-based commands
to make these and other changes.

Chris G4NAB

f5vmr, 2024-11-02:
These are of course in the svxlink, and not in the svxreflector. The
svxreflector is unconcerned with levels. It is up to the svxlink to
maintain correct levels in my opinion.
Chris

f5vmr, 2024-11-02:
Just to clarify my experience, I manage the svxportal-uk.ddns.net:81, and
have no need to monitor these levels at the reflector level. I have a node
that does that for me, by monitoring selected talkgroups. All the nodes
using the UK reflector seem to have balanced audio throughout.

Chris

cbspidi, 2024-11-03:
Hello, 
Apologies for not been clear with my initial message, yes there is reflector that i work with but the same hardware also serves a purpose of a local gateway access. 
Its the local gateway TX and RX audio line levels I would be interested to see, as @f5vmr  mentioned, this has nothing to do with reflector part of configuration.

I think a good way of implementing this feature would be to have a configurable "0/1" variable under below two section.
A simple suggestion would be `export_audio_level=0 or 1`, this would be very neat and handy feature.

```
[Rx1]
TYPE=Local
RX_ID=R
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0

[Tx1]
TYPE=Local
TX_ID=T
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
```

@f5vmr , tell me more about your config with alsa, what and how did you do it?

cbspidi, 2024-11-03:
Hi @f5vmr , i will ping you directly for more info on your way of configuring the svxliink, its always great to see how other engineers have it done.

On the feature request, it will be nice to have an input from someone who knows what would be involved in the code change for this to be implemented.

cbspidi, 2024-11-05:
Hello guys, 
It would be amazing to have this available via API call, but I understand there would need to be significant code implementations for the API system to be structured and thought about. 
Maybe it would be a good start for other parameters to be available and maybe changeable via API calls.

For now, if developers would be kind enough, it would be great to see this as an integer exported to the system variable

sm0svx, 2024-11-05:
The way to get your audio levels right is to use the devcal utility to calibrate all receivers (and transmitters) to exactly the correct level. That is, the received FM deviation on one receiver should ideally be reproduced on all other transmitters with exactly the same deviation.

But I agree that it would be useful for a reflector sysop being able to monitor audio levels for all nodes to find the nodes that does not have the correct calibration. I may add such a feature but I must be honest and say that it's not at the top of my TODO-list.

cbspidi, 2024-11-06:
Hello @sm0svx, 
It's great that you can see this has been very useful for sysop people. 
Going back to your TODO list, is there anything we can do as a temporary solution, something that is not perfect, or maybe not 100% accurate?

I was thinking of measuring peeks every 100ms; we already do it in the code to produce the warning message in the log.

A quick fix with a proper solution to be implemented later.

Im not a coder myself, but if you were to lead me in a direction of what and were would need to be changed for the quick fix I could go out and ask for help with other ham`s

cbspidi, 2024-11-13:
Hi Tobias, @sm0svx 

Would you have any comments on my last message?

cbspidi, 2025-03-02:
Hello Tobias @sm0svx , 
Checking to see if there anything we can assist with in pushing this request  forward?

cbspidi, 2025-03-11:
Hello, @sm0svx 

I would love to see this progress further; is there anything I could do to assist?
I'm not a coder, but if you were to lead me in the right direction, maybe I could make some changes, test them, and report back with the results.
What do you think?

sm0svx, 2025-03-16:
I'm sorry but I don't want to put any time on this right now. It will happen when I feel it's fun to work on.