sm0svx/svxlink #685: Svxlink problem running in read only FS in latest build

sa2blv, 2024-11-02:
The issue is that Svxlink attempts to change the ownership of a folder at startup. This operation is not permitted on a read-only file system.

-- Subject: A start job for unit svxlink.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support

--
-- A start job for unit svxlink.service has begun execution.
--
-- The job identifier is 1602.
Nov 02 22:06:11 SK3RIN sh[1109]: chown: changing ownership of '/var/lib/svxlink': Read-only file system
Nov 02 22:06:11 SK3RIN systemd[1]: svxlink.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStartPre= process belonging to unit svxlink.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Nov 02 22:06:11 SK3RIN systemd[1]: svxlink.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support

f5vmr, 2024-11-02:
In my experience there is no need to run svxlink in a read only system.
I have nodes that have run successfully for at least three years.
I am using Svxlink-server ver 24-02.1 svxlink v.1.8.0.

Chris g4nab

sa2blv, 2024-11-02:
Hi F5vmr, the problem is bigger when the nodes are located at sites that are only accessible some parts of the year due to deep snow or difficult-to-reach locations. In recent years, we have tried running repeaters 24/7 using ordinary SD cards, but some have failed. The read-only file system prevents Linux applications from writing repeatedly to sectors on the card, which increases stability for the embedded system.

f5vmr, 2024-11-03:
Ok understood. In 15 years of svxlink, I have had only two real failures,
but this has been to the user of inferior brands and speeds. A10 cards of
well known brands have been very successful. These days, I have tended to
use 16 gB cards.
It is certainly frustrating if a remote repeater suffers a failure.

At this time I use version 24.02-1 of svxlink-server, that has been
carefully archived in a repository accessible by a key, so that it can be
installed on bookworm by apt. This is not currently possible with the
ordinary apt as version 19-02 gets installed. The new version that is
available on the GitHub still requires compiling, and the security features
are still being appraised, but I do like the Talk-Group features and GPIOD
on ver.24, that are not available on version 19.
We have a couple of Swedish stations on svxportal-uk.ddns.net:81 that seem
not to fail.

73 Chris

sm0svx, 2024-11-04:
A note on secutity. You may also want to put the ca-bundle.crt file in /etc/svxlink/pki so that it does not have to be downloaded on each start (MITM-problem). It's not necessary but will increase security. You then also need to add the following to the configuration:
```
CERT_CAFILE=/etc/svxlink/pki/ca-bundle.crt
CERT_DOWNLOAD_CA_BUNDLE=0
```
The downside of making the CA bundle file read-only is that it cannot be updated by the reflector server. CA bundle updates should be very rare so it should not be a problem, even if your node is not reachable part of the year.

sm0svx, 2025-09-28:
Closing this issue since the suggested solution will resolve it

closed 2025-09-28 by sm0svx