sm0svx/svxlink #689: Need help to solve TX ON problem after svxlink stops

sp2ong, 2025-01-23:
Tobias, 

I have a request to add a small program
based on svxlink: https://github.com/sm0svx/svxlink/blob/master/src/svxlink/trx/PttHidraw.cpp which could be run independently, e.g.

**hidrawptt <device> <gpio> <status>**

e.g.

**hidrawptt /dev/hidraw0 3 0**

i.e. sets GPIO3 status LOW
(I don't know how in a situation when it is !GPIO3, i.e. operating ACTIVE_LOW - maybe add a parameter <active_low>?)

It would be very helpful in solving a problem that occurs. When SVXLINk has Tx running, i.e. PTT is ON  (there is a conversation oran announcement is being played) and at that time svxlink is stopped, e.g. systemctl stop svxlink or by shutdown without turning off the power supply from the computer, PTT remains in the HIGH state and the transmitter is constantly transmitting the carrier.
Such a program will help us enter it in svxlink.service in the line

**ExecStartPost=/usr/local/bin/hidrawptt 3 0**

set PTT to LOW state, i.e. turn it off after stopping svxlink

I am asking because I am not a programmer and I do not know C++ well and having the code in svxlink PttHidraw.cpp may be a small effort for you to add such a tool

While the same problem occurs when I use GPIO on the SBC bus, this problem is easily solved when using libgpiod using the gpioset program to set the GPIO PTT low line after finishing svxlink but in the case of Hidraw I do not have available programs

This will allow us to secure the PTT turn off of the transmitter when svxlink is stopped while transmitting, this applies to using Hidraw PTT among others

sm0svx, 2025-02-01:
That is a good idea to have some "PTT reset utility" in SvxLink.

It would be better though to not bind that utility to a specific type of PTT hardware so a higher level in the PTT code should be used, like using a Tx object created using the configuration from svxlink.conf.

sp2ong, 2025-02-01:
Tobias, it could probably be implemented in svxlink.conf but I'm looking for a solution to the problem using tool calls in svxlink.service where I have a script that checks the PTT bus type, if it's GPIOD so I can use the gpioset tool and if it's Hidraw, it currently has a python that changes the status but I don't like python because it's cool to implement but then it depends a lot on the version that's on the system

sp2ong, 2025-02-01:
Yes :-) nice solution

sp2ong, 2025-02-08:
Hi, Thank you for adding this solution.

I use in LADSAP plugin in Rx1 

When I put in svxlink.service 

ExecStopPost=/usr/bin/svxlink --config=${CFGFILE} --runasuser=${RUNASUSER} --reset --quiet

and stop svxlink when i Tx1 ON it is not works for me 

When I removed this line from svxlink.service and stop when svxlink has Tx1 ON after stopped svxlink PTT is still ON and run manually:

_/usr/bin/svxlink --config=/etc/svxlink/svxlink.conf --runasuser=svxlink --reset
SvxLink v1.8.99.15@24.02-74-gd7970497 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz

Starting logic: SimplexLogic
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/SimplexLogic.so
SimplexLogic: Loading RX "Rx1"
Rx1: Delay line (for DTMF muting etc) set to 220 ms
Rx1: "TAP Dynamics (M)" (tap_dynamics_m) by "Tom Szilagyi" (C) GPL
Rx1:     Path: /usr/lib/ladspa/tap_dynamics_m.so
Rx1:     In : "Attack [ms]" [4,500] = 4
Rx1:     In : "Release [ms]" [4,1000] = 650
Rx1:     In : "Offset Gain [dB]" [-20,20] = 8
Rx1:     In : "Makeup Gain [dB]" [-20,20] = 20
Rx1:     Out: "Envelope Volume [dB]" [-60,20] = -60
Rx1:     Out: "Gain Adjustment [dB]" [-60,20] = 0
Rx1:     In : "Function" [0,14] = 13
Rx1: "TAP Scaling Limiter" (tap_limiter) by "Tom Szilagyi" (C) GPL
Rx1:     Path: /usr/lib/ladspa/tap_limiter.so
Rx1:     In : "Limit Level [dB]" [-30,20] = -0.5
Rx1:     In : "Output Volume [dB]" [-30,20] = 0
Rx1:     Out: "latency" [0,2500.1] = 200
SimplexLogic: Loading TX "Tx1"
SimplexLogic: Loading module "ModuleParrot"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
	Module Parrot v1.1.1@24.02-74-gd7970497 starting...
SimplexLogic: Loading module "ModuleMeteoInfo"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
	Module Tcl v1.0.1@24.02-74-gd7970497 starting...
SimplexLogic: Loading module "ModuleMetarInfo"
	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
	Module MetarInfo v1.2.2@24.02-74-gd7970497 starting...
SimplexLogic: Event handler script successfully loaded.

Starting logic: ReflectorLogic
	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorV2Logic.so
ReflectorLogic: Connecting to service 
Activating link ReflectorLink
Initialization done. Exiting._

all works good

So I observed that LADSPA which I use during start svxlink need a few seconds about 3 to 5 second to load
When I removed LADSPA from configuration svxlink and put again in svxlink.service

ExecStopPost=/usr/bin/svxlink --config=${CFGFILE} --runasuser=${RUNASUSER} --reset --quiet

and test again all works

So it look like time loading configuration special LADSPA on SBC like I use Orange Pi Zero is break/not works this solution
I don't know it maybe we need optimize svxlink.service set time to longer allow run ExecStopPost program ?

sm0svx, 2025-02-08:
ok so if I understand correctly, the feature as implemented in SvxLink work but it is killed by systemd prematurely on timeout because LADSPA take too long to initialize?

sp2ong, 2025-02-08:
Yes it looks like this as you wrote

sm0svx, 2025-02-08:
Can you try to set the TimeoutStopSec= directive to some value, to see if it solves the problem?

sp2ong, 2025-02-08:
I  see during 

/usr/bin/svxlink --config=${CFGFILE} --runasuser=${RUNASUSER} --reset --quiet

is loading whole svxlink.conf typical like svxlink start normal. I understand that for --reset we need know konfiguration of GPIO used but it look like loading LADSPA need more time (I only use two plugins but maybe use more and another which need more  long time to loading)
it is obvious that the longer loading time of LADSPA in the --reset option in systemd is the problem. When I manually run it, the GPIO reset is OK

sp2ong, 2025-02-08:
Yes I found this and set TimeoutStopSec=30 but not help

sp2ong, 2025-02-08:
I try 60 but not help

sp2ong, 2025-02-08:
My svxlink.service
```

[Unit]
Description=SvxLink repeater control software
Documentation=man:svxlink(1)
After=network.target remote-fs.target syslog.target time.target

[Service]
Type=simple
Environment=RUNASUSER=svxlink
Environment=STATEDIR=/var/lib/svxlink/
Environment=CFGFILE=/etc/svxlink/svxlink.conf
Environment=LOGFILE=/var/log/svxlink
EnvironmentFile=/etc/default/svxlink
UMask=0002
ExecStartPre=/bin/touch ${LOGFILE}
ExecStartPre=/bin/chmod 0644 ${LOGFILE}
ExecStartPre=/bin/mkdir -m0775 -p ${STATEDIR}
ExecStartPre=/bin/sh -c 'chown ${RUNASUSER}:$(id -gn ${RUNASUSER}) "${STATEDIR}" "${LOGFILE}"'
ExecStart=/usr/bin/svxlink --logfile=${LOGFILE} --config=${CFGFILE} --runasuser=${RUNASUSER}
ExecStopPost=/usr/bin/svxlink --config=${CFGFILE} --runasuser=${RUNASUSER} --reset --quiet
TimeoutStopSec=30
ExecReload=/bin/kill -s HUP $MAINPID
Restart=on-failure
Nice=-10
TimeoutStartSec=60
TimeoutStopSec=10
LimitCORE=infinity
WorkingDirectory=/etc/svxlink

[Install]
WantedBy=multi-user.target
```

sm0svx, 2025-02-08:
Your have TimeoutStopSec set twice in there so the first one (30) maybe is overridden by the seconds one (10). But 10 seconds ought to be enough anyway I guess.

sp2ong, 2025-02-08:
Ah, you are right :-) 10 seconds for my LADSAP is to fast I need increase to 15 and now looks works OK
Thanks for point this mistake

sp2ong, 2025-02-08:
Hm, but still not always working, one time yes but next time not stop Tx1 PTT, I try set longer timer to 60 but it looks like LADSAP is problem during use --reset option

I suggest that this solution is need tested by more users

sp2ong, 2025-02-08:
In my log when I stop svxlink via systemd is

2025-02-08 18:16:34: ReflectorLogic: Using audio codec "OPUS"
2025-02-08 18:16:41: Tx1: Turning the transmitter OFF
2025-02-08 18:16:42: SimplexLogic: digit=6
2025-02-08 18:16:42: SimplexLogic: digit=0
2025-02-08 18:16:42: SimplexLogic: digit=#
2025-02-08 18:16:42: MeteoInfo: DTMF command received when idle: 0
2025-02-08 18:16:42: Playing WX Info
2025-02-08 18:16:42: **Tx1: Turning the transmitter ON**
2025-02-08 18:16:51: 
2025-02-08 18:16:51: SIGTERM received. Shutting down application...

sm0svx, 2025-02-08:
If you leave out the --quiet option you will get svxlink logs in your systemlog (journalctl). Maybe you can see something there. Or replace that option with --logfile=/tmp/svxlink-execstoppost.log temporarily.

sp2ong, 2025-02-08:
OK I will be try

My LADSAP linie in [Rx1] is

LADSPA_PLUGINS=tap_dynamics_m:4:650:8:20:13,tap_limiter:-0.5:0

sm0svx, 2025-02-08:
Maybe you can see in your logging what plugin it is that is taking a long time to start. Or if you have a lot of plugins installed that you don't use, maybe the indexing of LADSPA plugins are taking an unnecessarily long time.

If you type the `listplugins` command on the commandline, does that take a long time to execute?

Or `analyseplugin`, like `analyseplugin tap_dynamics_m`?

sp2ong, 2025-02-08:
Tobias, It works but sometimes it doesn't work but maybe it's the specifics of my computer etc.? I think if more people check it out for themselves especially those who use LADSPA

I will monitor the behavior on my system and if anything, please write, but I think we can close this issue.

Log show 

2025-02-08 18:28:13: SvxLink v1.8.99.15@24.02-74-gd7970497 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX
2025-02-08 18:28:13: 
2025-02-08 18:28:13: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
2025-02-08 18:28:13: welcome to redistribute it in accordance with the terms and conditions in the
2025-02-08 18:28:13: GNU GPL (General Public License) version 2 or later.
2025-02-08 18:28:13: 
2025-02-08 18:28:13: Using configuration file: /etc/svxlink/svxlink.conf
2025-02-08 18:28:13: --- Using sample rate 48000Hz
2025-02-08 18:28:13: 
2025-02-08 18:28:13: Starting logic: SimplexLogic
2025-02-08 18:28:13: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/SimplexLogic.so
2025-02-08 18:28:13: SimplexLogic: Loading RX "Rx1"
2025-02-08 18:28:13: Rx1: Delay line (for DTMF muting etc) set to 220 ms
2025-02-08 18:28:13: Rx1: "TAP Dynamics (M)" (tap_dynamics_m) by "Tom Szilagyi" (C) GPL
2025-02-08 18:28:13: Rx1:     Path: /usr/lib/ladspa/tap_dynamics_m.so
2025-02-08 18:28:13: Rx1:     In : "Attack [ms]" [4,500] = 4
2025-02-08 18:28:13: Rx1:     In : "Release [ms]" [4,1000] = 650
2025-02-08 18:28:13: Rx1:     In : "Offset Gain [dB]" [-20,20] = 8
2025-02-08 18:28:13: Rx1:     In : "Makeup Gain [dB]" [-20,20] = 20
2025-02-08 18:28:13: Rx1:     Out: "Envelope Volume [dB]" [-60,20] = -60
2025-02-08 18:28:13: Rx1:     Out: "Gain Adjustment [dB]" [-60,20] = 0
2025-02-08 18:28:13: Rx1:     In : "Function" [0,14] = 13
2025-02-08 18:28:13: Rx1: "TAP Scaling Limiter" (tap_limiter) by "Tom Szilagyi" (C) GPL
2025-02-08 18:28:13: Rx1:     Path: /usr/lib/ladspa/tap_limiter.so
2025-02-08 18:28:13: Rx1:     In : "Limit Level [dB]" [-30,20] = -0.5
2025-02-08 18:28:13: Rx1:     In : "Output Volume [dB]" [-30,20] = 0
2025-02-08 18:28:13: Rx1:     Out: "latency" [0,2500.1] = 200
2025-02-08 18:28:13: SimplexLogic: Loading TX "Tx1"
2025-02-08 18:28:13: SimplexLogic: Loading module "ModuleParrot"
2025-02-08 18:28:13: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
2025-02-08 18:28:13: 	Module Parrot v1.1.1@24.02-74-gd7970497 starting...
2025-02-08 18:28:13: SimplexLogic: Loading module "ModuleMeteoInfo"
2025-02-08 18:28:13: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
2025-02-08 18:28:13: 	Module Tcl v1.0.1@24.02-74-gd7970497 starting...
2025-02-08 18:28:13: SimplexLogic: Loading module "ModuleMetarInfo"
2025-02-08 18:28:13: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
2025-02-08 18:28:13: 	Module MetarInfo v1.2.2@24.02-74-gd7970497 starting...
2025-02-08 18:28:13: SimplexLogic: Event handler script successfully loaded.
2025-02-08 18:28:13: 
2025-02-08 18:28:13: Starting logic: ReflectorLogic
2025-02-08 18:28:13: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorV2Logic.so
2025-02-08 18:28:13: ReflectorLogic: Connecting to service 
2025-02-08 18:28:13: Activating link ReflectorLink
2025-02-08 18:28:13: Initialization done. Exiting.

sp2ong, 2025-02-08:
analyseplugin tap_dynamics_m 

this command show very fast result

but I observe when svxlink start is need about 5 to 8 second when displayed is information about LADSAP plugin

Rx1: "TAP Dynamics (M)" (tap_dynamics_m) by "Tom Szilagyi" (C) GPL
Rx1:     Path: /usr/lib/ladspa/tap_dynamics_m.so
Rx1:     In : "Attack [ms]" [4,500] = 4
Rx1:     In : "Release [ms]" [4,1000] = 650
Rx1:     In : "Offset Gain [dB]" [-20,20] = 8
Rx1:     In : "Makeup Gain [dB]" [-20,20] = 20
Rx1:     Out: "Envelope Volume [dB]" [-60,20] = -60
Rx1:     Out: "Gain Adjustment [dB]" [-60,20] = 0
Rx1:     In : "Function" [0,14] = 13
Rx1: "TAP Scaling Limiter" (tap_limiter) by "Tom Szilagyi" (C) GPL
Rx1:     Path: /usr/lib/ladspa/tap_limiter.so
Rx1:     In : "Limit Level [dB]" [-30,20] = -0.5
Rx1:     In : "Output Volume [dB]" [-30,20] = 0
Rx1:     Out: "latency" [0,2500.1] = 200

sm0svx, 2025-02-08:
If looking at your log, there does not seem to be anything taking extra time in your ExecStopPost. All log lines are on the same second.

sp2ong, 2025-02-08:
Yes you are right but it shows with an accuracy of one minute, I would have to make it show seconds

sp2ong, 2025-02-08:
Ah, it is in seconds but why when I see on screen  result log instead in to file I can count to 6 or 8 when is start display info about LADSAP  ? it like buffered

sm0svx, 2025-02-08:
I have no idea. Try to run it on the command like like: `date; svxlink --logfile=/tmp/svxlink.log --reset`. Is there a long time between the "date" printout and the time that end up in the log?

sp2ong, 2025-02-08:
Tobias, give more time to test, maybe during Sunday I will try to find any correlation with my system. If I found, I will be sending info. As I wrote it maybe it is only specified for my configuration  
I suggest close the issue and if anybody will report the problem you can reopen

sm0svx, 2025-02-08:
ok I'll close it. Please comment if you find out what it wrong.

closed 2025-02-08 by sm0svx

sp2ong, 2025-02-08:
Below result this command:

date; svxlink --logfile=/tmp/svxlink.log --reset

Sat 08 Feb 2025 19:05:17 PM CET

more /tmp/svxlink.log 

2025-02-08 19:05:28: SvxLink v1.8.99.15@24.02-74-gd7970497 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX
2025-02-08 19:05:28: 
2025-02-08 19:05:28: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
2025-02-08 19:05:28: welcome to redistribute it in accordance with the terms and conditions in the
2025-02-08 19:05:28: GNU GPL (General Public License) version 2 or later.
2025-02-08 19:05:28: 
2025-02-08 19:05:28: Using configuration file: /etc/svxlink/svxlink.conf
2025-02-08 19:05:28: --- Using sample rate 48000Hz
2025-02-08 19:05:28: 
2025-02-08 19:05:28: Starting logic: SimplexLogic
2025-02-08 19:05:28: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/SimplexLogic.so
2025-02-08 19:05:28: SimplexLogic: Loading RX "Rx1"
2025-02-08 19:05:28: Rx1: Delay line (for DTMF muting etc) set to 220 ms
2025-02-08 19:05:28: Rx1: "TAP Dynamics (M)" (tap_dynamics_m) by "Tom Szilagyi" (C) GPL
2025-02-08 19:05:28: Rx1:     Path: /usr/lib/ladspa/tap_dynamics_m.so
2025-02-08 19:05:28: Rx1:     In : "Attack [ms]" [4,500] = 4
2025-02-08 19:05:28: Rx1:     In : "Release [ms]" [4,1000] = 650
2025-02-08 19:05:28: Rx1:     In : "Offset Gain [dB]" [-20,20] = 8
2025-02-08 19:05:28: Rx1:     In : "Makeup Gain [dB]" [-20,20] = 20
2025-02-08 19:05:28: Rx1:     Out: "Envelope Volume [dB]" [-60,20] = -60
2025-02-08 19:05:28: Rx1:     Out: "Gain Adjustment [dB]" [-60,20] = 0
2025-02-08 19:05:28: Rx1:     In : "Function" [0,14] = 13
2025-02-08 19:05:28: Rx1: "TAP Scaling Limiter" (tap_limiter) by "Tom Szilagyi" (C) GPL
2025-02-08 19:05:28: Rx1:     Path: /usr/lib/ladspa/tap_limiter.so
2025-02-08 19:05:28: Rx1:     In : "Limit Level [dB]" [-30,20] = -0.5
2025-02-08 19:05:28: Rx1:     In : "Output Volume [dB]" [-30,20] = 0
2025-02-08 19:05:28: Rx1:     Out: "latency" [0,2500.1] = 200
2025-02-08 19:05:28: SimplexLogic: Loading TX "Tx1"
2025-02-08 19:05:28: SimplexLogic: Loading module "ModuleParrot"
2025-02-08 19:05:28: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
2025-02-08 19:05:28: 	Module Parrot v1.1.1@24.02-74-gd7970497 starting...
2025-02-08 19:05:28: SimplexLogic: Loading module "ModuleMeteoInfo"
2025-02-08 19:05:28: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
2025-02-08 19:05:28: 	Module Tcl v1.0.1@24.02-74-gd7970497 starting...
2025-02-08 19:05:28: SimplexLogic: Loading module "ModuleMetarInfo"
2025-02-08 19:05:28: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
2025-02-08 19:05:28: 	Module MetarInfo v1.2.2@24.02-74-gd7970497 starting...
2025-02-08 19:05:28: SimplexLogic: Event handler script successfully loaded.
2025-02-08 19:05:28: 
2025-02-08 19:05:28: Starting logic: ReflectorLogic
2025-02-08 19:05:28: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorV2Logic.so
2025-02-08 19:05:28: ReflectorLogic: Connecting to service 
2025-02-08 19:05:28: Activating link ReflectorLink
2025-02-08 19:05:28: Initialization done. Exiting

sp2ong, 2025-02-08:
Are is about 11-second difference between date time and time in log

when I remove LADSAP from config you can see is no delay between date and time in log

date; svxlink --logfile=/tmp/svxlink.log --reset

Sat 08 Feb 2025 19:12:59 CET

 more /tmp/svxlink.log 

2025-02-08 19:13:00: SvxLink v1.8.99.15@24.02-74-gd7970497 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX
2025-02-08 19:13:00: 
2025-02-08 19:13:00: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
2025-02-08 19:13:00: welcome to redistribute it in accordance with the terms and conditions in the
2025-02-08 19:13:00: GNU GPL (General Public License) version 2 or later.
2025-02-08 19:13:00: 
2025-02-08 19:13:00: Using configuration file: /etc/svxlink/svxlink.conf
2025-02-08 19:13:00: --- Using sample rate 48000Hz
2025-02-08 19:13:00: 
2025-02-08 19:13:00: Starting logic: SimplexLogic
2025-02-08 19:13:00: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/SimplexLogic.so
2025-02-08 19:13:00: SimplexLogic: Loading RX "Rx1"
2025-02-08 19:13:00: Rx1: Delay line (for DTMF muting etc) set to 220 ms
2025-02-08 19:13:00: SimplexLogic: Loading TX "Tx1"
2025-02-08 19:13:00: SimplexLogic: Loading module "ModuleParrot"
2025-02-08 19:13:00: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleParrot.so
2025-02-08 19:13:00: 	Module Parrot v1.1.1@24.02-74-gd7970497 starting...
2025-02-08 19:13:00: SimplexLogic: Loading module "ModuleMeteoInfo"
2025-02-08 19:13:00: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleTcl.so
2025-02-08 19:13:00: 	Module Tcl v1.0.1@24.02-74-gd7970497 starting...
2025-02-08 19:13:00: SimplexLogic: Loading module "ModuleMetarInfo"
2025-02-08 19:13:00: 	Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
2025-02-08 19:13:00: 	Module MetarInfo v1.2.2@24.02-74-gd7970497 starting...
2025-02-08 19:13:00: SimplexLogic: Event handler script successfully loaded.
2025-02-08 19:13:00: 
2025-02-08 19:13:00: Starting logic: ReflectorLogic
2025-02-08 19:13:00: 	Found plugin: /usr/lib/arm-linux-gnueabihf/svxlink/ReflectorV2Logic.so
2025-02-08 19:13:00: ReflectorLogic: Connecting to service 
2025-02-08 19:13:00: Activating link ReflectorLink
2025-02-08 19:13:00: Initialization done. Exiting.

so if I use LADSAP observe delay about 11 to 15 second so original TimeoutStopSec=10 was too short for my configuration ?

sp2ong, 2025-02-08:
but when I add --quiet  to run, not produce log file but with LADSAP time between start and finish is about 11 to 15 seconds

sp2ong, 2025-02-08:
it maybe we can skip reading/loading LADSAP when we use --reset because it is not necessary for this task, but I suppose you know better what can be done to optimize this.

sm0svx, 2025-02-08:
It's easy to say but not easy to implement since there is a need for a whole new mechanism in SvxLink to tell all components that we are just doing a reset and not a full startup so I'd rather not go that way. I have no idea why your LADSPA loading is so slow. On my RPi 3 it's fast.

sp2ong, 2025-02-08:
Ok Tobias, I have to deal with this problem :-) and I think that others do not have a similar situation so let's leave it and others use this option then after some time maybe experience will be gathered again thanks for your time and good luck

73 Waldek

sm0svx, 2025-02-08:
One way you could deal with it is to create a separate config file to use in the ExecStopPost run. You'd need to keep your different config files in sync manually but if this is a problem that only you have maybe that is acceptable.

sp2ong, 2025-02-09:
Thanks for the good tip and in fact in my case it was a good solution, I did reset.conf and it works very quickly :-)

```
[GLOBAL]
LOGICS=SimplexLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%Y-%m-%d %H:%M:%S"
CARD_SAMPLE_RATE=48000
CARD_CHANNELS=1

[SimplexLogic]
TYPE=Simplex
RX=Rx1
TX=Tx1
CALLSIGN=SP2ONG
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=pl_PL

[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:CARD=Codec,DEV=0
AUDIO_CHANNEL=0
SQL_DET=GPIOD
SQL_GPIOD_CHIP=gpiochip0
SQL_GPIOD_LINE=10

[Tx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:CARD=Codec,DEV=0
AUDIO_CHANNEL=0
PTT_TYPE=GPIOD
PTT_GPIOD_CHIP=gpiochip0
PTT_GPIOD_LINE=7

```