sm0svx/svxlink #690: SvxLink memory usage

ur3qjw, 2025-02-06:
Hi Tobias,

How much memory should SvxLink use?
Maybe SvxLink has memory problems?

I am using Debian 11 amd64.
Active: SimplexLogic, ReflectorLogic, QSO recorder.
Modules - disabled.
Is this normal behavior?

[image]
[image]
[image]
[image]

On the server, the numbers are generally impressive:
[image]

sm0svx, 2025-02-06:
Identifying memory issues may be hard. I don't know what systemctl get its memory usage information from but your "top" view looks normal.

You could try other tools if you want to look at memory usage over time. Try vmstat or sar (sysstat) to get system wide information. E.g. `vmstat 1` to continuously print memory usage or use the more advanced sysstat package which can record a range of system data every 10 minutes so that you can look at it later using the sar tool, like `sar -r`.

To continuously get the memory usage of svxlink try something like ` while ps --no-headers -C svxlink -ovsz,rss,%mem; do sleep 60; done | tee svxlink_memory.log` which will print/log the memory usage for SvxLink once every minute.

Or simply just use "top" and look at the memory usage for svxlink over time, remebering yourself what the previous reading was.

If any of these tools show that you are loosing memory over time, and that it never levels out, you may have identified a problem.

sm0svx, 2025-02-08:
Reopen the issue if you're pretty sure that there's a problem with SvxLink leaking memory

closed 2025-02-08 by sm0svx