sm0svx/svxlink #691: Telemetrie stoppd after install commit f9bd8c6

DL1GHN, 2025-02-08:
Hi,

the add of STATISTICS_INTERCAL=10 in svxlink.conf is made, svxlink did not send telemetry data to the APRS-IS after the update.

73, Pat, DL1GHN
ER-DB0LHR

sm3sgp, 2025-02-08:
Does it work if you remove the STATISTICS_INTERVAL parameter from svxlink.conf?
Did you use that parameter before the commit?

Can you share the [LocationInfo]  section?

DL1GHN, 2025-02-08:
Hi,
i don´t use the new parameter bevore the commit.
here´s my [Loctorinfo] Section - with the # in front of the STATISTICS_INTERVAL (for testing)

73..
[LocationInfo]
APRS_SERVER_LIST=euro.aprs2.net:14580
STATUS_SERVER_LIST=aprs.echolink.org:5199
LON_POSITION=7.51.44E
LAT_POSITION=48.20.55N
CALLSIGN=EL-DB0LHR
FREQUENCY=439.400
TX_POWER=8
ANTENNA_GAIN=6
ANTENNA_HEIGHT=10m
ANTENNA_DIR=-1
PATH=WIDE1-1
BEACON_INTERVAL=10
#STATISTICS_INTERVAL=10
TONE=1750
COMMENT=SvxLink by SM0SVX (www.svxlink.org)

DL1GHN, 2025-02-08:
It is working again with the deactivated STATISTICS_INTERVAL :-)

Thank you, Gunnar

sm3sgp, 2025-02-08:
I still there is a problem if you cannot set the STATISTICS_INTERVAL parameter. I have tested with changing the BEACON_INTERVAL and STATISTICS_INTERVAL, and I don't see any problems here. I am not using Echolink on my node.

sp2ong, 2025-02-09:
By the way, why do we have to specify EL or ER before the callsign when we want to send information about a station on aprs.fi and do not use echolink?

sp2ong, 2025-02-09:
It would be very useful to have the ability to send data only to aprs.fi with a callsign like SP2ABC-10
and send statistics to aprs.fi without having to send to STATUS_SERVER_LIST echolink server if I don't use EchoLink module. Telemetry/Statistics can also be sent to aprs.fi without having to use echolink server. Maybe ICON option can be added to set icon on aprs.fi ICON could be given to buoy STATION or REPEATER ??

sp2ong, 2025-02-09:
we define information in Locatioinfo about ANTENA an POWER but I don't see on aprs.fi this information it like PHG info is not sent ?

Should be like this on map?:

[image]

sm3sgp, 2025-02-09:
I think the ER and EL settings are from legacy Echolink, and I agree that it would be useful to use the connection to APRS-IS to send information about repeaters or simplex links reachable from RF.

A possibility set a different symbol would be good, but maybe a limited selection to some somewhat SvxLink related use cases, i.e Repeater.

What is shown on aprs.fi related to Telemetry and PHG might also be broken in some way, but unclear how.

sp2ong, 2025-02-09:
I see in svxlink code ANTENNA and POWER is used to send only to reflector server info via JSON but not to APRS

PHG frame which use ANTENNA and POWER info format is

https://www.aprs.org/aprsdos-pix/phg.txt
it maybe in future will nice add to send via APRS too

About CALLSIGN is I know that Echolink use only ER and EL but maybe we need know that ModuleEchoLink is defined and if not it allow use call sign like SP2ABC-10 where -19 example SSID and if not used ModuleEchoLInk is possible use define ICON on aprs.fi map like REPETER (tower: /r) or STATION (node : /n) 

But I realize that implementing this can take time and what I write is sometimes not easy to implement in code.

sp2ong, 2025-02-09:
But maybe there is an easier way to give the possibility to use the callsign type SP2ABC-SSID. If STATUS_SERVER_LIST= is not defined in Locationinfo and when it is defined the sign must be ER- EL-

sm0svx, 2025-02-11:
You're off-topic now so I'm closing this issue. If you want to discuss features it's better to use groups.io. (But I agree that the APRS code could need some rethinking)

I think that it's easy to be fooled that the statistics is not working because when SvxLink is restarted it always start with sequence number 000 and if you restart too often, aprs.fi will see the packets as duplicates since they have the same sequence number as one that was recently received.

closed 2025-02-11 by sm0svx