sm0svx/svxlink #695: SVXLINK should switch to using >=SigC++-3

dl1jbe, 2025-03-03:
At the moment the package still uses sigc++-2.xxx. That forces the build system to use an old C++ standard (-std=c++11) which is a main stumbling block for a port of `Qtel` to Qt6 as that will require a C++17 compiler (see Issue #686). 

Switching to sigc++-3.xxx would  allow to use the more modern C++17.

f5vmr, 2025-03-03:
As I and some others generally build without QT in the CMAKE command, this
has no effect even using 24.02. but maybe in the future this will have to
be considered.

sm0svx, 2025-03-04:
It's a bit of work to switch to libsigc++3 but doable. It has to be done sooner or later so it's more of a matter of how long older systems should be supported. I'd like to make one more release with C++11 and then move forward. But I'm not quite ready to make a new release just yet.

dl1jbe, 2025-03-07:
Thanks for the given time frame.

sm0svx, 2025-03-10:
Partly working build for sigc++3 in [branch sigc++](https://github.com/sm0svx/svxlink/pull/new/sigc++3).

dl1jbe, 2025-03-18:
While not complete yet, at least the part needed to compile `qtel` seems to be ok. That allows me to start to work on the Qt6 port.

sm0svx, 2025-09-28:
Sigc++3 and Qt6 support now is in git master

closed 2025-09-28 by sm0svx

dl1jbe, 2025-09-29:
Great. Thanks. Any timeline for a new release? Otherwise I will prepare a patched version for Gentoo so we can move away from qt5.

sm0svx, 2025-09-29:
A release is at least a couple of weeks away and maybe more. Depends on how much free time I have to work on this project. So I cannot give any promises.