sm0svx/svxlink #697: SVXReflector <-> EchoLink bridge not working

YO3LNK, 2025-03-15:
Hi,
I'm making a svx network with nodes and i also want to connect a EchoLink server via svxlink. I upgraded to latest version (svxLink v1.8.99.35@24.02). My client connects to svxreflector but no traffic is going through.

svxlink.conf:
```
[GLOBAL]
LOGICS=SimplexLogic,ReflectorLogic
CFG_DIR=svxlink.d
# TIMESTAMP_FORMAT="%c"
TIMESTAMP_FORMAT="%Y-%m-%d %H:%M:%S"
LINKS=DXLink
CARD_CHANNELS=1
CARD_SAMPLE_RATE=48000

[SimplexLogic]
TYPE=Simplex
RX=NONE
TX=TXStream
MODULES=ModuleEchoLink
CALLSIGN=YO3KEX
SHORT_IDENT_INTERVAL=5
LONG_IDENT_INTERVAL=10
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=ro_RO
RGR_SOUND_DELAY=0
RGR_SOUND_ALWAYS=1
# REPORT_CTCSS=123
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
# MUTE_RX_ON_TX=0
# MUTE_TX_ON_RX=0
# MACROS=Macros

[ReflectorLogic]
TYPE=ReflectorV2
MODULES=ModuleEchoLink
# DNS_DOMAIN=dxlink.ro
HOSTS=192.168.1.68:5300     
HOST_PORT=5300
CALLSIGN=DXLink      
AUTH_KEY="xxx"
OPUS_ENC_BITRATE=32000
DEFAULT_TG=226
MONITOR_TGS=226
# NODE_INFO_FILE=/etc/svxlink/node_info.json
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=ro_RO
JITTER_BUFFER_DELAY=0
MUTE_FIRST_TX_LOC=0
MUTE_FIRST_TX_REM=0
TMP_MONITOR_TIMEOUT=3600

[DXLink]
CONNECT_LOGICS=SimplexLogic,ReflectorLogic
DEFAULT_ACTIVE=1
# TIMEOUT=0

# [MultiTx]
# TYPE=Multi
# TRANSMITTERS=Tx1

[TXStream]
TYPE=Local
AUDIO_DEV=alsa:hw:Loopback,0,2
AUDIO_CHANNEL=0
PTT_TYPE=NONE
TIMEOUT=7200
TX_DELAY=0
PREEMPHASIS=0
```
svxreflector.conf:
```
[GLOBAL]
TIMESTAMP_FORMAT="%c"
LISTEN_PORT=5300
CERT_CA_HOOK=/usr/share/svxlink/ca-hook.py
TG_FOR_V1_CLIENTS=226
HTTP_SRV_PORT=8080
ACCEPT_CALLSIGN=".*"

# [ReflectorLogic]
# TYPE=Reflector
# HOSTS=0.0.0.0:5300
# CALLSIGN=YO3LNK-L
# AUTHORIZED_IPS=*
# MAX_CONNECTIONS=100
# AUTH_KEY=yo3kex69
# EVENT_HANDLER=/usr/local/share/svxlink/events.tcl

[USERS]
DXLink="YO3KEX-L"
YO3LNK-P="YO3LNK-P"

[PASSWORDS]
YO3KEX-L="xxx"
YO3LNK-P="xxx"

[ROOT_CA]
#KEYFILE=svxreflector_root_ca.key
#CRTFILE=svxreflector_root_ca.crt
#COMMON_NAME=SvxReflector Root CA
#ORG_UNIT=SvxLink
#ORG=MyOrg
#LOCALITY=MyTown
#STATE=StateOrProvince
#COUNTRY=XX
#EMAIL_ADDRESS=sysop@svxlink.example.org

[ISSUING_CA]
#KEYFILE=svxreflector_issuing_ca.key
#CSRFILE=svxreflector_issuing_ca.csr
#CRTFILE=svxreflector_issuing_ca.crt
#COMMON_NAME=SvxReflector Issuing CA
#ORG_UNIT=SvxLink
#ORG=MyOrg
#LOCALITY=MyTown
#STATE=StateOrProvince
#COUNTRY=XX
#EMAIL_ADDRESS=sysop@svxlink.example.org

[SERVER_CERT]
#KEYFILE=svxreflector.key
#CSRFILE=svxreflector.csr
#CRTFILE=svxreflector.crt
COMMON_NAME=dxlink.ro
#ORG_UNIT=SvxLink
#ORG=MyOrg
#LOCALITY=MyTown
#STATE=StateOrProvince
#COUNTRY=XX
#SUBJECT_ALT_NAME=DNS:public-hostname.example.org,IP:172.17.1.42
#EMAIL_ADDRESS=sysop@svxlink.example.org

# [USERS]
#SM0ABC-1=MyNodes
#SM0ABC-2=MyNodes
#SM3XYZ=SM3XYZ

# [PASSWORDS]
#MyNodes="Change this key now!"
#SM3XYZ="A strong password"

#[TG#9999]
#AUTO_QSY_AFTER=300
#ALLOW=S[A-M]\\\\d.*|LA8PV
#SHOW_ACTIVITY=0
```

If you need anything, tell me.
YO3LNK Stefan

f5vmr, 2025-03-15:
Firstly svxreflector is a separate entity and should run in its own device
although it can run in your svxlink machine but not recommended.
The [ReflectorLogic] should then be configured to be able to connect to
this new entity, and have the callsign and Auth_key that should be also in
the [Users] and [Passwords] in the SvxReflector.conf in the running
instance of svxreflector.
Svxreflector is another application within the Svxlink Package, and runs to
accept all other svxlink instances that are similarly configured to link
with it. As it happens at the moment you only have an svxlink instance
running.
Pleased to help you during your configuration.
Chris G4NAB

YO3LNK, 2025-03-15:
So the problem is that it runs on the same machine.
I also have a node that connects and with the main echolink svxlink. When I TX on echolink nothing happens on the reflector, but the clients are connected, you can see them in the list.
I'm going to try to run them on diffrent devices and see, although I prefer to run them on same host. Let me know if that is possible or I must run on diffrent servers.

f5vmr, 2025-03-15:
I correct my previous mail, as I see that you combined both your svxlink
and svxreflector confs on the same page, which was confusing. But I see
some commands that is I think incompatible "accept_callsign" and
listen_port do not appear in my svxlink system in the reflector.
Also TG_For_V1_Clients is for an earlier version of svxlink prior to
version 24.... Version 19 which is used in the French Network.
My only other query would be, have you set up ModuleEchoLink.conf?

YO3LNK, 2025-03-15:
Yes ModuleEchoLink is set up. I can connect via phone, just isn't tx-ing on reflector, just on client, although they are connected. 
I'm removing TG_For_V1_Clients.
PS: Accept Callsign is a value in the new svx that blocks callsigns without numbers or the correct format (Ex. DXLink blocked, only YO3KEX allowed or simillar...).

Version 24 seems to me a bit to messy, but is used by Romanian main Network and I tried it. Maybe i'll switch to 19, enough for what I need to do.

ModuleEchoLink.conf:

```
[ModuleEchoLink]
NAME=EchoLink
ID=2
#ALLOW_IP=192.168.1.0/24
#DROP_INCOMING=^()$
#REJECT_INCOMING=^()$
#ACCEPT_INCOMING=^(.*)$
#REJECT_OUTGOING=^()$
#ACCEPT_OUTGOING=^(.*)$
#REJECT_CONF=0
#CHECK_NR_CONNECTS=2,300,120
SERVERS=europe.echolink.org
CALLSIGN=YO3LNK-L
PASSWORD=Frantic0506@
SYSOPNAME=Stefan
LOCATION=ONLY TESTING!
#PROXY_SERVER=the.proxy.server
#PROXY_PORT=8100
#PROXY_PASSWORD=PUBLIC
#BIND_ADDR=10.20.30.40
# MAX_QSOS=10
# MAX_CONNECTIONS=11
# LINK_IDLE_TIMEOUT=300
#AUTOCON_ECHOLINK_ID=9999
#AUTOCON_TIME=1200
#USE_GSM_ONLY=1
DEFAULT_LANG=ro_RO
#COMMAND_PTY=/dev/shm/echolink_ctrl
REMOTE_RGR_SOUND=1
DESCRIPTION="YO3KEX DXLink"
# DESCRIPTION="You have connected to a SvxLink node,\n"
# 	    "a voice services system for Linux with EchoLink\n"
# 	    "support.\n"
# 	    "Check out http://svxlink.sf.net/ for more info\n"
# 	    "\n"
# 	    "QTH:     My_QTH\n"
# 	    "QRG:     Simplex link on ???.??? MHz\n"
# 	    "CTCSS:   My_CTCSS_fq_if_any Hz\n"
# 	    "Trx:     My_transceiver_type\n"
# 	    "Antenna: My_antenna_brand/type/model\n"

```

f5vmr, 2025-03-15:
So I now understand fully what you have done. This is an independent
radio-free bridge. I suspect that the loopback soundcard may not be
performing as required.
Perhaps reconstruct the[Rx1] and [Tx1] using the loopback card parameters
in AUDIO_DEV with the appropriate TYPE=LOCAL,
and for Rx1 SQL=NONE and for Tx1 PTT_TYPE=DUMMY.
To see if that performs better.
Chris

YO3LNK, 2025-03-15:
Yes, that's right radio-free for now. I also have radio nodes but not using them at the moment. I take it in steps :)
I'm rewriting tx and rx and I'll let you know

f5vmr, 2025-03-15:
Version 19 isn't talkgroup compatible, so stick with 24. I use 24.2.1 that
has no SSL certification, and works very well with all connections here
including Tetra. The Echolink connections are left to individual nodes, but
during their use, the full svxreflector is not in service for that node
until that Echolink call is terminated. As far as our network is concerned,
Echolink is tertiary and almost consigned to legacy use only. We have
bridges to other domains that far outweigh the use for Echolink now.

Portal.svxlink.uk:81 is our portal view.

Chris

YO3LNK, 2025-03-15:
Ok so I did a bit of more testing and the problem is that it doesn't enter the reflector because of the RX setting not configured.
I programmed the Rx1 function but no "none/dummy" option like in TX. I'll try with VOX.
The purpose of this client only to make a bridge(a link) between the main reflector(the network) and the EchoLink clients

YO3LNK, 2025-03-15:
Maybe I need for this to use a RepeaterLogic, not simplex logic.

Stefan

f5vmr, 2025-03-15:
Yes I understand completely.

My EchoLink is unique to my Node only, so that if anyone calls, my link is passively disconnected from the network, using the command MUTE_LOGIC_LINK=1 which in your ModuleEchoLink.conf you do not have.

You on the other hand wish the connection to pass to the network for everyone to hear. Be aware that this option may be contrary to the EchoLink.org <http://echolink.org/> rules as they do not wish to have multi web-like connections to their servers.

But you won't be alone I’m sure.

I think VOX should work quite well as a Squelch option. Although you will have to examine the VOX Levels.

If I can help in any way, you only have to ask.

73
Chris G4NAB

We have

f5vmr, 2025-03-15:
Simplex Logic should be adequate. Adding RepeaterLogic would complicate it unnecessarily.

Chris

YO3LNK, 2025-03-15:
In YO all the networks are also bridged to EchoLink, it's like a kind of standard :)
Other SQL_DET option besides VOX I don't know if it'll function. 
The echolink svxlink server will be on it's own server and this is it's only purpose.
I'll try to set the VOX to open as you press tx on echolink, to be sensible enough.

Stefan YO3LNK

sp2ong, 2025-03-15:
In your ModuleEchoLin.conf you are missing one parameters

MUTE_LOGIC_LINKING=1

or

MUTE_LOGIC_LINKING=0

depending on whether you want to send it to the network or only via radio EchoLink traffic

and try to replace

CONNECT_LOGICS=SimplexLogic,ReflectorLogic

to

CONNECT_LOGICS=SimplexLogic:9:SVX,ReflectorLogic

YO3LNK, 2025-03-15:
I used "MUTE_LOGIC_LINKING=0" without needing more RX or TX modifications. Thank you so much Chris and Waldek. I really appreciate your much needed help.
That was the solution, the missing parameter.
I hope this helps anyone who also encounters this issue.

All the best, 73 good weekend!
Stefan YO3LNK (13 yrs old)

closed 2025-03-15 by YO3LNK