sm0svx/svxlink #699: Macros can only start with "D"

keithtweed, 2025-03-24:
Defined macros are all given DTMF commands prefixed with `d`, which is hard-coded in SVXLink. Many commercial radios do not support entry of 4th column DTMF digits (`ABCD`), either from the front panel or altogether. It would be nice if macros were not forced to have a prefix, and instead incoming DTMF was evaluated against the list of currently defined macros to determine if it is a valid macro. Barring this, allowing the prefix to be configured would also solve this issue. Thank you!

f5vmr, 2025-03-24:
Can I ask which radios of which you speak?

Even Tytera/Retevis/Baofeng/Quanshen do have the full range of DTMF  keys
although in the case of Tytera/Retevis they are unmarked in respect of A-D.
I believe it though I've not tested it myself that the green menu key has
to be pushed first. I'll give it a test this morning myself. I have Yaesu
radios as well as all the above, and I've a couple of dead Motorola radios,
an ICOM,  with full keypads.
Annoyingly I've a Retevis RT73 with a full key pad that the developers
failed to install a DTMF tone generator/or tone activator profile in the
CPS.

As inconvenient as it might be, there are smart phone apps that generate
tones in sequences that could assist, when held against the microphone when
transmitting.

In my experience, It will not be possible to initiate a DTMF sequence with
any other key, due to the possibility that ALL the numbers and the * key
can be used to commence other module functions, if they were initiated.
That leaves only A-D available.

keithtweed, 2025-03-25:
The two I have found so far are Motorola and Harris, which lack support for `ABCD` altogether - both on the front panel, and in the CPS. We want our repeater system to be compatible with 12-key DTMF devices, hence the issue. 

SVXLink could take the list of defined macros and check for these explicitly rather than dedicating a prefix to macro commands. This solves compatibility with 12-key DTMF devices, and avoids interfering with other modules unless a macro command is explicitly defined with an entire sequence used by another command.

sm0svx, 2025-05-03:
I think this is a good idea. Adding a MACRO_PREFIX configuration variable which defaults to D would keep backwards compatibility. Setting it to empty would allow matching macros from the start of the entered DTMF command.