sm0svx/svxlink #700: QTel and svxlink echo link on the same network

jawilson104, 2025-03-27:
I can't find this issue on the issue board.  Simple question: can QTEL operate on the same network as svxlink EchoLink?  If you are using the PC EchoLink client from Echolink.org, it won't work because of an issue with sockets.  Thanks for your help.

f5vmr, 2025-03-27:
In some installations QTel is not installed, and certainly the case with
stations in the UK network.

f5vmr, 2025-03-27:
I'll qualify that by saying my installations in general are Debian 12
based, and never windows.

sm0svx, 2025-05-02:
Please use the svxlink group at groups.io for asking questions.

You have to use an Echolink proxy since the Echolink protocol only work for one client per IP.

closed 2025-05-02 by sm0svx