sm0svx/svxlink #702: Trying to reconnect to APRS server

dl7ata, 2025-04-06:
The message _Trying to reconnect to APRS server_ once seems to be normal, when running the system over a long time.
Today I found 
`06.04.2025 17:14:47: *** WARNING: Disconnected from APRS server
06.04.2025 17:14:52: *** WARNING: Trying to reconnect to APRS server
06.04.2025 17:14:55: *** WARNING: Disconnected from APRS server
06.04.2025 17:15:00: *** WARNING: Trying to reconnect to APRS server  ...
` in the log.
for more than 20 minutes. Then I restart svxlink and the system was suddenly connected to an APRS server.

dl7ata, 2025-04-14:
Since one week, it was the only event and not reproducible. So I'll close it.

sm0svx, 2025-04-14:
I believe there may be a bug with APRS reconnect so I'll look into it

sm0svx, 2025-04-29:
Thanks for reporting again. Can you look back in your log to see if you can find the point where this started. Can you see an error or warning message that started the reconnect cycle?  Something else that looks suspicious?

sm0svx, 2025-04-29:
Also find the previous successful connect to see what the IP was. Was it the same as the one you had above?
What host have you specified in your LocationInfo/APRS_SERVER_LIST config?
What was the duration of the problem this time?  That is, when did it start and when did you restart SvxLink?

dl7ata, 2025-04-29:
- Nothing suspicous, no error, no warning:

>29.04.2025 12:00:00: 2mTx: Turning the transmitter ON
29.04.2025 12:00:17: 2mTx: Turning the transmitter OFF
29.04.2025 12:00:25: 70cmTx: Turning the transmitter OFF
29.04.2025 12:05:51: *** WARNING: Disconnected from APRS server
29.04.2025 12:05:56: *** WARNING: Trying to reconnect to APRS server
...
`
- last successful connect was:

> 29.04.2025 11:10:09: *** WARNING: Disconnected from APRS server
29.04.2025 11:10:14: *** WARNING: Trying to reconnect to APRS server
29.04.2025 11:10:14: Connected to APRS server 109.72.122.50 on port 14580

grep on "APRS" on svxlink-log:
>29.04.2025 12:10:02: *** WARNING: Trying to reconnect to APRS server
29.04.2025 12:10:10: *** WARNING: Disconnected from APRS server
29.04.2025 12:10:15: *** WARNING: Trying to reconnect to APRS server
29.04.2025 12:10:15: Connected to APRS server 109.72.122.50 on port 14580
29.04.2025 12:10:21: SimplexLogic: Loading module "ModuleAPRSMsg"
29.04.2025 12:10:21: Connected to APRS server 51.210.15.145 on port 14580
29.04.2025 13:05:06: *** WARNING: Disconnected from APRS server
29.04.2025 13:05:11: *** WARNING: Trying to reconnect to APRS server
29.04.2025 13:05:11: Connected to APRS server 85.188.1.129 on port 14580
29.04.2025 13:08:04: *** WARNING: Disconnected from APRS server
29.04.2025 13:08:09: *** WARNING: Trying to reconnect to APRS server
29.04.2025 13:08:09: Connected to APRS server 85.188.1.129 on port 14580
29.04.2025 14:07:09: *** WARNING: Disconnected from APRS server
29.04.2025 14:07:14: *** WARNING: Trying to reconnect to APRS server
29.04.2025 14:07:14: Connected to APRS server 57.128.11.1 on port 14580
29.04.2025 14:07:44: *** WARNING: Disconnected from APRS server
29.04.2025 14:07:49: *** WARNING: Trying to reconnect to APRS server
29.04.2025 14:07:49: Connected to APRS server 57.128.11.1 on port 14580
`
Host: APRS_SERVER_LIST=euro.aprs2.net:14580
Duration today was 5 minutes, i.e. that I restarted 5 minutes after the first error in log.

Digging in log, I found another one from today. Started at 
29.04.2025 11:07:17: *** WARNING: Disconnected from APRS server
and ended
29.04.2025 11:10:14: *** WARNING: Trying to reconnect to APRS server
29.04.2025 11:10:14: Connected to APRS server 109.72.122.50 on port 14580 

Current Log started 27.04.2025 06:31:42. So today seems to be the only events.

sm0svx, 2025-04-29:
Ok thanks!

One theory is that the DNS lookup end up picking a node that is not reachable at the moment when the reconnect is initiated. Since the TTL (time-to-live) for euro.aprs2.net is five minutes (300 seconds), a new host will not be picked until after five minutes. You restarted SvxLink after just under five minutes. If it happens again, let it run to see if it will reconnect.

That behavior is due to a sub-optimal DNS implementation is SvxLink, so that is something to work on. On each reconnect try it should choose a random server from the list returned by the DNS lookup.

sm0svx, 2025-04-29:
Try the latest master commit (5a79ca1) which fix the DNS problem described above. Hopefully it will resolve your issue.

dl7ata, 2025-04-30:
Thanks Tobias. Compiling is running and I'll let you know my experience.

dl7ata, 2025-05-01:
I got this phenomenon today:   

>01.05.2025 08:05:53: *** WARNING: Disconnected from APRS server 85.90.180.26:14580: Connection closed by remote peer
01.05.2025 08:05:58: Trying to reconnect to APRS server...
01.05.2025 08:05:58: Connected to APRS server 109.72.122.50 on port 14580
01.05.2025 08:07:55: *** WARNING: Disconnected from APRS server 109.72.122.50:14580: Connection closed by remote peer
01.05.2025 08:08:00: Trying to reconnect to APRS server...
01.05.2025 08:08:00: Connected to APRS server 195.190.142.207 on port 14580
01.05.2025 08:08:35: *** WARNING: Disconnected from APRS server 195.190.142.207:14580: Connection closed by remote peer
01.05.2025 08:08:40: Trying to reconnect to APRS server...
01.05.2025 08:08:40: Connected to APRS server 86.123.190.5 on port 14580
01.05.2025 08:09:17: *** WARNING: Disconnected from APRS server 86.123.190.5:14580: Connection closed by remote peer
01.05.2025 08:09:22: Trying to reconnect to APRS server...
01.05.2025 08:09:22: Connected to APRS server 85.90.180.26 on port 14580
01.05.2025 08:09:58: *** WARNING: Disconnected from APRS server 85.90.180.26:14580: Connection closed by remote peer
01.05.2025 08:10:03: Trying to reconnect to APRS server...
01.05.2025 08:10:04: Connected to APRS server 109.72.122.50 on port 14580
01.05.2025 08:10:39: *** WARNING: Disconnected from APRS server 109.72.122.50:14580: Connection closed by remote peer
01.05.2025 08:10:44: Trying to reconnect to APRS server...
01.05.2025 08:10:44: Connected to APRS server 86.123.190.5 on port 14580
01.05.2025 08:11:55: *** WARNING: Disconnected from APRS server 86.123.190.5:14580: Connection closed by remote peer
01.05.2025 08:12:00: Trying to reconnect to APRS server...
01.05.2025 08:12:00: Connected to APRS server 109.72.122.50 on port 14580
01.05.2025 08:12:42: *** WARNING: Disconnected from APRS server 109.72.122.50:14580: Connection closed by remote peer
01.05.2025 08:12:47: Trying to reconnect to APRS server...
01.05.2025 08:12:47: Connected to APRS server 51.83.187.31 on port 14580
01.05.2025 08:13:23: *** WARNING: Disconnected from APRS server 51.83.187.31:14580: Connection closed by remote peer
01.05.2025 08:13:28: Trying to reconnect to APRS server...
01.05.2025 08:13:29: Connected to APRS server 34.57.210.69 on port 14580
01.05.2025 08:13:59: *** WARNING: Disconnected from APRS server 34.57.210.69:14580: Connection closed by remote peer
01.05.2025 08:14:04: Trying to reconnect to APRS server...
01.05.2025 08:14:04: Connected to APRS server 34.57.210.69 on port 14580'

08:14:04 last error message after successful connect.

dl7ata, 2025-05-02:
Just checked the log:
Since yesterday until now (17:15) , I got 53 
_*** WARNING: Disconnected from APRS server <sevral_IPs>:14580: Connection closed by remote peer_

sm0svx, 2025-05-02:
What you are experiencing feel more like a networking problem rather than a bug in SvxLink. I have no idea what else it could be. At least SvxLink now try to connect to a random host in the range of IPs returned by the DNS lookup. That SvxLink recover by itself after a while is pointing to that SvxLink is doing its job.