sm0svx/svxlink #703: Segfault during audio processing

petrkr, 2025-04-25:
svxlink v1.8.0@24.02 segfaults during audio processing

```
FT991: Turning the transmitter ON
svxlink: /home/petrkr/.cache/pikaur/build/svxlink/src/svxlink-24.02/src/async/audio/AsyncAudioProcessor.cpp:322: void Async::AudioProcessor::writeFromBuf(): Assertion `(written >= 0) && (written <= buf_cnt)' failed.
Aborted (core dumped)
```

results is forever keyed radio

```
[1822787.672034] svxlink[1184950]: segfault at 5711c4b8b9e7 ip 00007f103e99146a sp 00007ffdeacda4a0 error 4 in libsigc-2.0.so.0.0.0[346a,7f103e991000+2000] likely on CPU 0 (core 0, socket 0)
[1822787.672043] Code: 48 0f 6e c0 48 c7 40 10 00 00 00 00 66 0f 6c c0 c6 40 18 00 48 89 03 0f 11 00 5b c3 0f 1f 40 00 f3 0f 1e fa 55 53 48 83 ec 08 <48> 8b 1f c6 47 18 01 48 39 df 74 3c 48 89 fd 0f 1f 80 00 00 00 00
```

petrkr, 2025-04-25:
Now it crashed again and I have some "better" stack trace

```
Apr 25 20:56:34 pkdell systemd-coredump[9849]: [🡕] Process 2393 (svxlink) of user 1000 dumped core.
                                               
                                               Stack trace of thread 2393:
                                               #0  0x000076444e877624 n/a (libc.so.6 + 0x97624)
                                               #1  0x000076444e81dba0 raise (libc.so.6 + 0x3dba0)
                                               #2  0x000076444e805582 abort (libc.so.6 + 0x25582)
                                               #3  0x000076444e8054eb n/a (libc.so.6 + 0x254eb)
                                               #4  0x000076444f1dd1ca n/a (libasyncaudio.so.1.7 + 0x1e1ca)
                                               #5  0x000076444f1e1699 _ZN5Async14AudioProcessor12writeSamplesEPKfi (libasyncaudio.so.1.7 + 0x22699)
                                               #6  0x00005c3a6b859d14 _ZN9PeakMeter12writeSamplesEPKfi (/usr/bin/svxlink + 0xe4d14)
                                               #7  0x000076444f1e1305 _ZN5Async14AudioProcessor12writeFromBufEv (libasyncaudio.so.1.7 + 0x22305)
                                               #8  0x000076444f1e1699 _ZN5Async14AudioProcessor12writeSamplesEPKfi (libasyncaudio.so.1.7 + 0x22699)
                                               #9  0x000076444f1f2076 _ZN5Async9AudioFifo20writeSamplesFromFifoEv (libasyncaudio.so.1.7 + 0x33076)
                                               #10 0x000076444f1f22d9 _ZN5Async9AudioFifo12writeSamplesEPKfi (libasyncaudio.so.1.7 + 0x332d9)
                                               #11 0x00005c3a6b7fc66a _ZN5Async10AudioValve12writeSamplesEPKfi (/usr/bin/svxlink + 0x8766a)
                                               #12 0x000076444f1ed6fc _ZN5Async11AudioDevice9putBlocksEPsm (libasyncaudio.so.1.7 + 0x2e6fc)
                                               #13 0x000076444f2098cc _ZN5Async15AudioDeviceAlsa16audioReadHandlerEPNS_7FdWatchEt (libasyncaudio.so.1.7 + 0x4a8cc)
                                               #14 0x000076444f20a961 _ZN5Async15AudioDeviceAlsa9AlsaWatch9readEventEPNS_7FdWatchE (libasyncaudio.so.1.7 + 0x4b961)
                                               #15 0x000076444f230752 _ZN5Async14CppApplication4execEv (libasynccpp.so.1.7 + 0xe752)
                                               #16 0x00005c3a6b7e19bc main (/usr/bin/svxlink + 0x6c9bc)
                                               #17 0x000076444e807488 n/a (libc.so.6 + 0x27488)
                                               #18 0x000076444e80754c __libc_start_main (libc.so.6 + 0x2754c)
                                               #19 0x00005c3a6b7e3425 _start (/usr/bin/svxlink + 0x6e425)
                                               
                                               Stack trace of thread 9847:
                                               #0  0x000076444e87dbe2 n/a (libc.so.6 + 0x9dbe2)
                                               #1  0x000076444e871e33 n/a (libc.so.6 + 0x91e33)
                                               #2  0x000076444e871e74 n/a (libc.so.6 + 0x91e74)
                                               #3  0x000076444e8ec53e __poll (libc.so.6 + 0x10c53e)
                                               #4  0x000076444d9e2a78 n/a (libpulse.so.0 + 0x32a78)
                                               #5  0x000076444d9cc654 pa_mainloop_poll (libpulse.so.0 + 0x1c654)
```

sm0svx, 2025-05-02:
Can you try the latest development version of SvxLink to see if you get the same error?

petrkr, 2025-05-02:
Bad is, last segfault I saw in logs at 29th April and since that nope..  I have uptime 7 days, so reboot does not "solved" issue...

I really love such random errors which I can not reproduce. I maybe changed some settings since that, like disabled roger beeping or added sounds.

but most often it caused mine Android 12 echolink app, sometimes right after first transmit, sometimes it "survived" like half hour. But for now it was without segfault for 4 days.

Maybe it is connected with some glitch on pulseaudio subsystem if there is opened some other app which use different sound card, because now there is no other app using pulseaudio (like brave browser or so)

First I want to be sure I am able reproduce, then I can say "yes master branch does not do that anymore"

sm0svx, 2025-05-03:
I don't recommend running SvxLink with Pulseaudio enabled but if you have to, expect problems. It looks like a bug so if we can fix it, it would be good. But I don't think it's a common problem among SvxLink-users. The better solution would of course be adding support for Puleseaudio in SvxLink and that may happen some day but is not prioritized.

petrkr, 2025-05-03:
Yes, but todays systems use something like that, pulseaudio or pipewire.

For now I have svxlink as remote link for radio. At same computer I have digitals like fldigi, wsjtx etc and nothing had problem with pulseaudio backend.

It's enough if svxlink will not segfault, but just print some warning about missing samples or something like that, but do not crash.

I'll try to find out reproduce, today I stressed it for hour, connecting, disconnecting, tx, rx and I can not reproduce...  but I think Murphy will bring it when there will be some meeting in air and I will want to participate.

I'll just monitor it and let you know if I'll find something more useful.

petrkr, 2025-07-21:
Today it occurs again

`a41fcf6f (tag: 25.05.1, origin/maint) Prepare for release 25.05.1`

FT991: Turning the transmitter ON
svxlink: /home/petrkr/git/svxlink/src/async/audio/AsyncAudioProcessor.cpp:322: void Async::AudioProcessor::writeFromBuf(): Assertion `(written >= 0) && (written <= buf_cnt)' failed.
Aborted (core dumped)