sm0svx/svxlink #705: echolib_test --logon

iddq, 2025-05-03:
Hi, I tried to execute a logon with the test program, but it didn't work. I have the feeling that it's trying an old-fashioned login method with a plain text password. Can you tell me how to fix this? Thank you.

sm3sgp, 2025-05-03:
Please add more information on what you are doing?
What software are you trying to use? What version?
What is your configuration?

iddq, 2025-05-03:
compiled from git.

Well, I don't know if it was a temporary issue or something else, so I'll describe the situation in detail:

First, I tried this: echolib_test --logon

Unfortunately, this doesn't provide any feedback, even though it would be very useful to see whether the login was successful or not.

Later, I added the --stations option.
I tried running it several times, but I always got an error saying:
"The directory server closed the connection before all data was received."
That's when I started investigating, because I suspected that the logon might not have been successful.

I noticed that the Android EchoLink client already uses challenge-response authentication, and it seems that unfortunately svxlink does not.
Although this wasn't the cause of the current issue, it might become a problem in the future.

After that, I experimented with different passwords in the EchoLink client, just to observe how it behaves.
Then, once I logged in again with the correct password, the svxlink test program started working as well.
Of course, the password in the test program's configuration remained unchanged the entire time.

I can also imagine that the issue was with server1, which is hardcoded in the test program.

sm0svx, 2025-05-03:
You can use the "-v" switch to make the application more verbose.

Yes SvxLink use plaintext passwords. Do you have any more information on how the challenge/response authentication is implemented in Echolink?  Protocol spec?

iddq, 2025-05-21:
Unfortunately, I don't have any information about that. Do you use any kind of chat for development-related communication? Telegram, IRC?