sm0svx/svxlink #706: Segfault when ModuleEchoLink reads the SERVER= line

F5VCV, 2025-05-13:
Hi,  I’m getting a segfault when I activate ModuleEcholink.
Environment :  SvxLink [v1.9.0@25.05](mailto:v1.9.0@25.05)  compiled on Debian 12 (bookworm) running on an RPi3 and an RPi4.
Interestingly, I don’t get a segfault if the line SERVERS=  is commented out, although a config without that line doesn’t male sense, it may help in understanding what’s happening.

I’ve added a gdb trace and backtrace.
(gdb) r
Starting program: /usr/local/bin/svxlink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
SvxLink v1.9.0@25.05 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /usr/local/etc/svxlink/svxlink.conf
--- Using sample rate 16000Hz

Starting logic: RepeaterLogic
	Found plugin: /usr/local/lib/svxlink/RepeaterLogic.so
RepeaterLogic: Loading RX "Rx1"
Rx1: Delay line (for DTMF muting etc) set to 75 ms
RepeaterLogic: Loading TX "Tx1"
RepeaterLogic: Loading module "ModuleHelp"
	Found /usr/local/lib/svxlink/ModuleHelp.so
	Module Help v1.0.0@25.05 starting...
RepeaterLogic: Loading module "ModuleParrot"
	Found /usr/local/lib/svxlink/ModuleParrot.so
	Module Parrot v1.1.1@25.05 starting...
RepeaterLogic: Loading module "ModuleEchoLink"
	Found /usr/local/lib/svxlink/ModuleEchoLink.so
	Module EchoLink v1.6.0@25.05 starting...

Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7d9dfd8 in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, sock=-1, 
    remote_addr=..., remote_port=0, recv_buf_len=1024, __in_chrg=<optimized out>)
    at /opt/sources/svxlink/src/async/core/AsyncTcpConnection.cpp:198
198	  : remote_addr(remote_addr), remote_port(remote_port), sock(sock)
(gdb) bt
#0  0x0000007ff7d9dfd8 in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, sock=-1, 
    remote_addr=..., remote_port=0, recv_buf_len=1024, __in_chrg=<optimized out>)
    at /opt/sources/svxlink/src/async/core/AsyncTcpConnection.cpp:198
#1  0x0000007ff7d9de14 in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, 
    recv_buf_len=549610610688, __in_chrg=<optimized out>)
    at /opt/sources/svxlink/src/async/core/AsyncTcpConnection.cpp:174
#2  0x0000007ff65eaa24 in EchoLink::DirectoryCon::DirectoryCon(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, Async::IpAddress const&) () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#3  0x0000007ff65dbc98 in EchoLink::Directory::createClientObject() () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#4  0x0000007ff65defb4 in EchoLink::Directory::Directory(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Async::IpAddress const&) () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#5  0x0000007ff66756e8 in ModuleEchoLink::initialize (this=0x5555a53d70)
    at /opt/sources/svxlink/src/svxlink/modules/echolink/ModuleEchoLink.cpp:347
#6  0x00000055557af2c4 in Logic::loadModule (this=0x55559abec0, module_cfg_name="ModuleEchoLink")
    at /opt/sources/svxlink/src/svxlink/svxlink/Logic.cpp:1395
#7  0x00000055557aea60 in Logic::loadModules (this=0x55559abec0)
    at /opt/sources/svxlink/src/svxlink/svxlink/Logic.cpp:1305
#8  0x00000055557ab7ac in Logic::initialize (this=0x55559abec0, cfgobj=..., logic_name="RepeaterLogic")
    at /opt/sources/svxlink/src/svxlink/svxlink/Logic.cpp:626
#9  0x0000007ff6aeddac in RepeaterLogic::initialize (this=0x55559abec0, cfgobj=..., logic_name="RepeaterLogic")
    at /opt/sources/svxlink/src/svxlink/svxlink/RepeaterLogic.cpp:163
#10 0x000000555579a6a0 in initialize_logics (cfg=...) at /opt/sources/svxlink/src/svxlink/svxlink/svxlink.cpp:786
#11 0x0000005555799760 in main (argc=1, argv=0x7ffffffc68) at /opt/sources/svxlink/src/svxlink/svxlink/svxlink.cpp:517
(gdb)

F5VCV, 2025-05-13:
NOTE: The previous gdb log may have come from source with some comments I had inserted.
This one is from a fresh git clone ... so only svxlink source

(gdb) r
Starting program: /usr/local/bin/svxlink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
SvxLink v1.9.0@25.05 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /usr/local/etc/svxlink/svxlink.conf
--- Using sample rate 16000Hz

Starting logic: RepeaterLogic
	Found plugin: /usr/local/lib/svxlink/RepeaterLogic.so
RepeaterLogic: Loading RX "Rx1"
Rx1: Delay line (for DTMF muting etc) set to 75 ms
RepeaterLogic: Loading TX "Tx1"
RepeaterLogic: Loading module "ModuleHelp"
	Found /usr/local/lib/svxlink/ModuleHelp.so
	Module Help v1.0.0@25.05 starting...
RepeaterLogic: Loading module "ModuleParrot"
	Found /usr/local/lib/svxlink/ModuleParrot.so
	Module Parrot v1.1.1@25.05 starting...
RepeaterLogic: Loading module "ModuleEchoLink"
	Found /usr/local/lib/svxlink/ModuleEchoLink.so
	Module EchoLink v1.6.0@25.05 starting...

Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7d9decc in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, sock=-1, 
    remote_addr=..., remote_port=0, recv_buf_len=549610610688, __in_chrg=<optimized out>)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/async/core/AsyncTcpConnection.cpp:164
164	  : remote_addr(remote_addr), remote_port(remote_port), sock(sock)
(gdb) bt
#0  0x0000007ff7d9decc in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, sock=-1, 
    remote_addr=..., remote_port=0, recv_buf_len=549610610688, __in_chrg=<optimized out>)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/async/core/AsyncTcpConnection.cpp:164
#1  0x0000007ff7d9de0c in Async::TcpConnection::TcpConnection (this=0x5555a58220, __vtt_parm=0x400, 
    recv_buf_len=549610610688, __in_chrg=<optimized out>)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/async/core/AsyncTcpConnection.cpp:157
#2  0x0000007ff65eaa24 in EchoLink::DirectoryCon::DirectoryCon(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, Async::IpAddress const&) () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#3  0x0000007ff65dbc98 in EchoLink::Directory::createClientObject() () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#4  0x0000007ff65defb4 in EchoLink::Directory::Directory(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Async::IpAddress const&) () from /lib/aarch64-linux-gnu/libecholib.so.1.3
#5  0x0000007ff6675708 in ModuleEchoLink::initialize (this=0x5555a53d70)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/modules/echolink/ModuleEchoLink.cpp:347
#6  0x00000055557af304 in Logic::loadModule (this=0x55559abec0, module_cfg_name="ModuleEchoLink")
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/Logic.cpp:1395
#7  0x00000055557aeaa0 in Logic::loadModules (this=0x55559abec0)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/Logic.cpp:1305
#8  0x00000055557ab7ec in Logic::initialize (this=0x55559abec0, cfgobj=..., logic_name="RepeaterLogic")
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/Logic.cpp:626
#9  0x0000007ff6aeddcc in RepeaterLogic::initialize (this=0x55559abec0, cfgobj=..., logic_name="RepeaterLogic")
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/RepeaterLogic.cpp:163
#10 0x000000555579a6e0 in initialize_logics (cfg=...)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/svxlink.cpp:786
#11 0x00000055557997a0 in main (argc=1, argv=0x7ffffffc68)
    at /home/user/DO_NOT_USE_just_for_source_compare/svxlink/src/svxlink/svxlink/svxlink.cpp:517
(gdb)

sm0svx, 2025-05-13:
I have not tested SvxLink on 64 bit ARM så there may be bugs. Try to use a 32 bit OS for now. It will probably be a while until this is fixed.

F5VCV, 2025-05-15:
Thank You Tobias.
Is works on the 32bit OS.

sp2ong, 2025-05-16:
I updated on computer which uses **64-bit** Debian 11 svxlink to version 1.9 and ModuleEchoLink stopped working:
 
2025-05-16 07:11:37: SvxLink v1.9.0@... Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX
2025-05-16 07:11:37: 
....
2025-05-16 07:11:37: Starting logic: SimplexLogic
2025-05-16 07:11:37:     Found plugin: /usr/lib/x86_64-linux-gnu/svxlink/SimplexLogic.so
....
2025-05-16 07:11:37:     Found /usr/lib/x86_64-linux-gnu/svxlink/ModuleEchoLink.so
2025-05-16 07:11:37:     Module EchoLink v1.6.0@... starting...
2025-05-16 07:11:37: SimplexLogic: Event handler script successfully loaded.
 
Show on logs ERROR:
 
2025-05-16 07:13:37: *** ERROR: Command timeout while communicating to the directory server
 
 
When I switch to 1.8.99.38  version svxlink
 
2025-05-16 07:19:52: SvxLink v1.8.99.38@... Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX
 
Module EchoLink connects to the directory server without problem:
 
2025-05-16 07:19:53: EchoLink directory status changed to ON
2025-05-16 07:19:55: --- EchoLink directory server message: ---
2025-05-16 07:19:55: EchoLink Server v2.6.159
2025-05-16 07:19:55: 
 
ModuleEchoLink don't use a proxy server and has set SERVER=servers.echolink.org

sp2ong, 2025-05-16:
I added info that platform is x86_64 GNU/Linux  not ARM and it looks like on 64 bits there are problems with ModuleEchoLink