sm0svx/svxlink #713: Message after stop svxlink: static LinkManager* LinkManager::instance(): Assertion `_instance != 0' failed

sp2ong, 2025-08-06:
Hi

I have used latest svxlink code 

2025-08-06 18:41:34: SvxLink v1.9.99.22@25.05.1-35-g6b56a264 Copyright (C) 2003-2025 Tobias Blomberg / SM0SVX

When I run the command:

systemctl stop svxlink 

in the log file, show the message in the last line

2025-08-06 18:41:59: Tx1: Turning the transmitter ON
2025-08-06 18:42:00: Tx1: Turning the transmitter OFF
2025-08-06 18:42:03: 
2025-08-06 18:42:03: NOTICE: SIGTERM received. Shutting down application...
2025-08-06 18:42:03: ReflectorLogic: Disconnected from 192.168.1.1:5300: Locally ordered disconnect
2025-08-06 18:42:03: svxlink: /opt/src/svxlink/src/svxlink/svxlink/LinkManager.h:181: static LinkManager* LinkManager::instance(): Assertion `_instance != 0' failed.

closed 2025-08-15 by sp2ong

sm0svx, 2025-08-15:
I think this still is a problem. Did it get resolved by itself on your node?

f5vmr, 2025-08-16:
Hi Tobias
I’m running a couple of nodes on V25 (TYPE=ReflectorV2) with our V24 Reflector, and have had no issues at all.
73
Chris